renode/renode #572: Support for tflm on nucleo-l4r5zi

jsncepu, 2024-01-01:
Hi. I used x-cube-ai package in stm32cubeide to generate a small tflm application (e.g., sine model).  It can run normally on a real nucleo-l4r5zi board. I can see the output in lpuart1 from a virtual serial port. However, when I tried to load the generated elf file into renode (I used the demo repl and resc of the officicial : nucleo_l4r5zi-hello_world.resc and nucleo_l4r5zi-hello_world.repl), I got nothing from lpusart1. It seems that the official demo uses zephyr rtos. Is it the problem affecting my port effeorts? How can I get a barel version of repl for l4r5zi which doesn't include zephyr rtos just like stm32f746-bb-repl?  Could anynoe can help me?  Thank you very much!

mateusz-holenko, 2024-01-18:
Hi @jsncepu, thanks for asking a question.

Judging from the names of `resc` and `repl` scripts I assume you've found them on the [Dashboard](https://zephyr-dashboard.renode.io/) or [Renodepedia,](https://renodepedia.renode.io/boards/nucleo_l4r5zi/), is that correct? In such case yes, those are based on the Zephyr RTOS.

Renode is software agnostic, meaning that it should be able to run any payload that works on actual HW (ranging from custom baremetal, through RTOSes like Zephyr, more advanced OSes like Linux, etc.). The level of support might vary between platforms, though - it depends on the completeness of models.

In order to debug what's happening with the simulation I suggest enabling the following logs in Monitor/resc script:
```
cpu0 LogFunctionNames true
sysbus LogPeripheralAccess lpuart1 true
```
It might be that your software loops during initialization before it even reaches any UART output, but this should be visible in logs.

BTW, if you're interested in tflm, we also have a running demo of it in https://renodepedia.renode.io/boards/nucleo_l4r5zi/?view=software&demo=tensorflow_lite_micro (but it's also Zephyr based).

jsncepu, 2024-01-22:
Thank you for your reply @mateusz-holenko . I will try to debug my program according to your advice. I guess the main problem is that I don't know how to configure the parameters in the repl file for baremetal board without the RTOS becuase when I run my problem directly on the example based on the example resc and repl files, the monitor of Renode reported the error of no vector table. However, after I commentted out the line of codes of vector table, I got nothing from lpuart port.

PiotrZierhoffer, 2024-02-01:
The `vector_table` line is Zephyr-specific. We typically try to guess the vector table offset for Cortex-M, but it doesn't work on every binary. You can set it manually to a symbol that is available in your ELF:

```
cpu0 VectorTableOffset `sysbus GetSymbolAddress "your_symbol_name"`
```

Do you see a reasonable output when you run the `LogFunctionNames` command mentioned above? Does it seem like your binary is actually executing?