renode/renode #576: Zephyr STM32F4/F0 usart1 output

axibo-reiner, 2024-01-11:
### Description

I am expecting to get an output on usart1 when the machine is configured in the way described below. However nothing is displayed. I am attempting this on both the F4_disco and F072_disco boards and getting the same result. The built binary works on real hardware

### Expected behaviour

kernel log printed on terminal

### How to reproduce?

To get the `.elf`, also changing to printk from printf. This is what is in the zephyr samples repo, usart1 is enabled in the dts and set as the shell/console output in the alias
`west build -b stm32f072b_disco samples/hello_world`

```
mach create
machine LoadPlatformDescription @platforms/boards/stm32f072b_discovery.repl 
sysbus LogPeripheralAccess sysbus.usart1
logLevel -1 sysbus.usart1
showAnalyzer sysbus.usart1
sysbus LoadELF @/{PATHTOELF}
```

```
18:23:32.1759 [INFO] Loaded monitor commands from: /opt/renode/scripts/monitor.py
18:23:51.1854 [INFO] System bus created.
18:24:04.3404 [INFO] sysbus: Logging already enabled: False.
18:25:38.6181 [INFO] sysbus: Loading segment of 12472 bytes length at 0x8000000.
18:25:38.6315 [INFO] sysbus: Loading segment of 120 bytes length at 0x80030B8.
18:25:38.6316 [INFO] sysbus: Loading segment of 4 bytes length at 0x8003130.
18:25:38.6316 [INFO] sysbus: Loading segment of 4216 bytes length at 0x20000000.
18:26:10.1025 [INFO] cpu: Guessing VectorTableOffset value to be 0x8000000.
18:26:10.1179 [INFO] cpu: Setting initial values: PC = 0x80003D1, SP = 0x20001078.
18:26:10.1218 [INFO] machine-0: Machine started.
18:26:10.1766 [WARNING] sysbus: [cpu: 0x8000256] (tag: 'SYSCFGCOMP') ReadDoubleWord from non existing peripheral at 0x40010000, returning 0x00000000.
18:26:10.1774 [WARNING] sysbus: [cpu: 0x8000256] (tag: 'SYSCFGCOMP') WriteDoubleWord to non existing peripheral at 0x40010000, value 0x3.
18:26:10.2460 [WARNING] sysbus: [cpu: 0x8001502] (tag: 'FLASH_INTERFACE') ReadDoubleWord from non existing peripheral at 0x40022000, returning 0x00000000.
18:26:10.2461 [WARNING] sysbus: [cpu: 0x8001502] (tag: 'FLASH_INTERFACE') WriteDoubleWord to non existing peripheral at 0x40022000, value 0x1.
18:26:10.2462 [WARNING] sysbus: [cpu: 0x8001502] (tag: 'FLASH_INTERFACE') ReadDoubleWord from non existing peripheral at 0x40022000, returning 0x00000000.
18:26:10.2462 [WARNING] sysbus: [cpu: 0x800151E] (tag: 'FLASH_INTERFACE') ReadDoubleWord from non existing peripheral at 0x40022000, returning 0x00000000.

```

### Environment

Please, provide the following information:

- Ubuntu 22 x86
- Renode, version 1.14.0.29506 (81da46c1-202308081623)

### Additional information

Please add all the information you feel is relevant but not clearly visible in the reproduction repo or otherwise.

### Do you plan to address this issue and file a PR?

No, i am still new to renode and am still trying to build the base knowledge

johannesgensheimer, 2025-01-28:
@axibo-reiner have you been able to resolve that problem? Currently having the exact same issue.

axibo-reiner, 2025-01-28:
@johannesgensheimer unfortunately I was not able to. I ended up using the native simulation tool and it worked really well. 

https://docs.zephyrproject.org/latest/boards/native/native_sim/doc/index.html

Then using `socat -d -d pty,raw,echo=0 pty,raw,echo=0` to create some virtual /tty interfaces that could be used for host communication to the host software.

johannesgensheimer, 2025-01-28:
All right, thanks @axibo-reiner !