renode/renode #578: Using renode API in C#

asmitaj08, 2024-01-16:
Hi,
The renode doc shows how we can access its APIs like Start, setHooks, etc. either via executing these within that renode terminal or passing it though a .resc or a python script. Is there any doc , which exposes its APIs that can be natively used within any c# script?
Also, is the breakpoint set possible only via gdb? 
Thank you!!

PiotrZierhoffer, 2024-03-21:
Hi @asmitaj08 

I assume that by C# script you mean .resc scripts? They are not strictly C#, as they also support our [internal Python engine](https://renode.readthedocs.io/en/latest/basic/using-python.html) and have their own syntax/grammar .

Unfortunately, it's almost impossible to describe the whole API, as we try to expose the whole interface - it's not hand written, but rather autogenerated.

For now I have to refer you to our [documentation](https://renode.readthedocs.io/en/latest/) and  [scripts](https://github.com/renode/renode/tree/master/scripts) for samples.

Regarding breakpoints, look for `AddHook` in https://renode.readthedocs.io/en/latest/basic/using-python.html#cpu-hooks

A short snippet would be `cpu AddHook 0xYOURADDRESS "self.Pause()"`

And then resume with `cpu Resume`.

Having said that, GDB interface is typically easier to use.

mateusz-holenko, 2024-03-21:
You can also use hook to change execution mode of the CPU at given address:
```
cpu AddHook 0x80009000 "self.ExecutionMode = Antmicro.Renode.Peripherals.CPU.ExecutionMode.SingleStepNonBlocking"
```

This will make CPU stop at 0x80009000 and switch to the single stepping mode. Now you can use `cpu Step` to iterate over instructions.

closed 2024-07-04 by PiotrZierhoffer