renode/renode #580: STM32_Timer AutoReload should only be 16-bits

xobs, 2024-01-18:
### Description

The STM32_Timer.cs definition has a field for AutoReload that is 32-bits:

https://github.com/renode/renode-infrastructure/blob/0d02f3a70ac2eab91992718caf7198906cc1076c/src/Emulator/Peripherals/Peripherals/Timers/STM32_Timer.cs#L340

However, the actual field appears to be only 16-bits:

https://github.com/renode/renode-infrastructure/blob/0d02f3a70ac2eab91992718caf7198906cc1076c/src/Emulator/Peripherals/Peripherals/Timers/STM32_Timer.cs#L340

As a result, writing `0xffff` to this register gets sign-extended and ends up waiting effectively forever.

PiotrZierhoffer, 2024-02-01:
Hi @xobs, thanks for spotting this. Can you please correct the second link? I assume it was for something else, like ref manual? I'm asking because it might be a difference between versions of the SoC (we reuse this model a lot, maybe this should be a config option for a SoC).

xobs, 2024-02-01:
Apologies -- I've updated the Issue with an image taken from RM0090 which shows the STM32F429. I've seen a similar issue on the STM32H7, which is similarly 16-bits.

PiotrZierhoffer, 2024-02-01:
Thanks a lot

MateuszKarlic, 2024-02-08:
Hi @xobs
Since, we reuse this peripheral model with 16- and 32-bit configuration (e.g. on `stm32h753`, timer2 is 32 bit, while timer3 is 16 bit), we opted to clamp fields at registration time, by inferring the timer's fields width from the value of `initialLimit` (https://github.com/renode/renode-infrastructure/commit/519d5146d71b1d763d770a11c4e822985b564cb4), with the only exception being the Prescaler, that always should be 16 bits wide. So, if you set the `initialLimit` to lower value, it will clamp the length of these fields, as for example is the case here: 

https://github.com/renode/renode/blob/f55c2c03977067501dabfb76273e6dc1aa585cfa/platforms/cpus/stm32f4.repl#L153-L156

The model will select the minimal width, where the `initialLimit` will fit within the timer's registers - so for limit of 0xFFFF, 16-bit registers will be generated.

Thanks for reporting this issue! Please check if this fix works for you.

xobs, 2024-02-08:
That's a subtle catch, thanks for finding the `initialLimit` issue!

The fix works for me.

closed 2024-02-08 by xobs