renode/renode #586: STMG0 UART DMA Tx/Rx

ItamarEliakim-R, 2024-02-12:
### Description

Feature Request:
I am trying to implement UART DMA TX/RX for the STM32G0B1.
For some reason, the DMA doesn't trigger the DMA callback functions. 
```
HAL_UART_RxCpltCallback
```
Or
``` 
DMA1_Ch4_7_DMA2_Ch1_5_DMAMUX1_OVR_IRQHandler
```

### Usage example
Minimal example of STM32G0B1, using STM32CubeMX.
With TX/RX DMA enabled on UART1/UART2.

### Additional information
Under certain infrastructure modifications, I see that the IRQ is triggered for the RX functions, meaning that the 
``` 
USART1_IRQHandler
``` 
Is triggered properly.
The issues arise when working/testing the DMA functions, which are defined by startup_stm32g0xx.s.

### Do you plan to address this issue and file a PR?
Yes

mateusz-holenko, 2024-03-21:
Hi @ItamarEliakim-R, thanks for opening the issue.

Could you create a CI setup based on our [reproduction template](https://github.com/renode/renode-issue-reproduction-template)?

We provide a support for STM32 DMA controller models in Renode already, so it might be a matter of properly configuring the platform.

tomhepworth, 2024-10-31:
Was there any update to this? I am struggling to get the zephyr async uart API to work nicely with the STM32F4's DMA

JohnZhang3IoT, 2025-03-11:
> Was there any update to this? I am struggling to get the zephyr async uart API to work nicely with the STM32F4's DMA
> 
> To get RX to work I had to add the DMA to the list of UART interrupts
> 
> ```
> usart2: UART.STM32_UART @ sysbus <0x40004400, +0x100>
>     IRQ -> nvic@38 | dma1@5 
> ```
> 
> However RX misses bytes when the double buffer swaps over, and the uart API callbacks do not seem to work as intended.

Hi @tomhepworth , I met same issue, my firmware is using uart idle interrupt + DMA to receive the data. In real hardware stm32f4, it can handle the data whenever I send bytes from serial monitor, but in renode, it can only trigger DMA when the receive buffer full, e.g: I defined uart receive buffer size is 64, and if I send 63 bytes, the idle call back won't trigger unless I send one more bytes next time to reach the size 64.
Any update of your work? thanks.