renode/renode #589: robot-keywords.robot not able to find the executable for Linux

admorgan, 2024-02-16:
### Description

I integrate Renode into CI through using the Robot Framework keywords. 1.13.3 and 1.14.0 releases do not work with this infrastructure because they can not find the Renode.exe executable. I am able to use all prior versions. 

### Expected behaviour

Importing renode-keywords.robot into my tests should give me access to renode and allow me to create machines to test against.

Instead I get an output that it can't find binary: 

> Robot Framework remote server binary not found (/opt/renode/tests/../output/bin/Release/Renode.exe). Did you forget to build it in Release configuration?

### Environment

Please, provide the following information:

- OS: Ubuntu 22.04
- Renode version (with commit SHA): 1.14.0 release both portable and .deb

### Do you plan to address this issue and file a PR?

If you let me know if it the binary needs to be moved in the release or should I change the path I would be happy to make a PR.

admorgan, 2024-02-16:
To clarify if I change the value of ${DIRECTORY} to ${CURDIR}/unit-tests/RenodeTests/bin/${CONFIGURATION} everything works as expected.

admorgan, 2024-02-19:
We are using renode to test our code in CI via the renode-keywords.robot. This is an ultra simplified version of what we do, it isn't complete enough to run, but should give you an idea.

```
*** Settings ***
Library    RobotSocketUtil.py
Setup    Suite Setup
Teardown    Suite Teardown

*** Test Cases ***
Run Emulated Hardware Test
[Setup]    Test Setup
[Teardown]    Test Teardown
   Execute Command    mach create
    Execute Command    machine LoadPlatformDescription@test.repl
    Execute Command    sysbus LoadELF @app.axf
    Execute Command    sysbus ApplySVD @test.svd
    Execute Command    sysbus.uart0 CreateFileBackend @test.out
    Start Emulation
    Wait for line on uart    </testsuites>

*** Keywords ***
Suite Setup
    Build app
    ${RENODE_PORT_NUMBER} =    RobotSocketUtil.get_free_port
    Set Global Variable   ${RENODE_PORT_NUMBER}    ${RENODE_PORT_NUMBER}
    Set Global Variable    ${DISABLE_XWT}    ${TRUE}
    Import Resource    ${RENODE_DIR}/tests${/}/renode-keywords.robot
    renode-keywords.Setup

Suite Teardown
    renode-keywords.Teardown

Test Setup
    Import Library    Remote    http://127.0.0.1:${PORT_NUMBER}
    Reset Emulation

Test Teardown
    Run Keyword and Ignore Error    End UART Capture
    Reset Emulation
```

mateusz-holenko, 2024-03-21:
@admorgan, thanks for more details.

FTR, starting Renode with the use of `renode-keywords.robot` is not an option we normally use (this logic is now handled by the `renode-test` script).

Having said that, setting `{DIRECTORY}` (and possibly `{RENODETOOLS}`) variables is the right solution to the problem.
The default values corresponds to paths layout in the source code and I think it could be improved when generating packages.

closed 2025-07-03 by PiotrZierhoffer