renode/renode #590: Arm64 host

urielka, 2024-03-05:
First of all I would like to say thank you for this amazing project!

I am working on a Linux computer running an arm64 processor, from what I understand the emulator part of renode is written in C/C++ so I need to build it from source.

I started the build and noticed errors around inline asm that looked like x86, so my question is do you support running on a arm64 host? Do I need to change a flag?

PiotrZierhoffer, 2024-03-07:
Hi @urielka 

While we do intend to enable arm64 host, it's not yet officially supported - we rely on binary translation and arm64 requires a different backend to be implemented.

On macOS  (M1/M2) we have reports of working well on the Rosetta emulation layer, but Linux is not there yet.

urielka, 2024-03-07:
Thanks, makes sense!