renode/renode #598: CAN Bus Example

yassinedevop, 2024-04-17:
I want to simulate a CAN Bus communication using socket CAN from the host and the target is an STM32F4 emulated on renode, can anyone suggest an example to start things out.

dzid26, 2024-04-30:
Example of flexcan https://github.com/renode/renode/blob/master/tests/peripherals/S32K3XX_FlexCAN.robot.  There are other tests with CAN too.

Seems like F0 and H7 are the only ones with CANbus currently. 

https://github.com/renode/renode/blob/d02cf7521df7d7d728668f50d9cccbdb917ccb13/platforms/cpus/stm32f0.repl#L167
https://github.com/renode/renode/blob/d02cf7521df7d7d728668f50d9cccbdb917ccb13/platforms/cpus/stm32h743.repl#L280

But STCAN is implemented [here](https://github.com/renode/renode-infrastructure/blob/master/src/Emulator/Peripherals/Peripherals/CAN/STMCAN.cs) so maybe it is just a matter of just adding few lines to repl for F4 board (or STM32F103 in my case).

yassinedevop, 2024-05-01:
Thanks for the help but my question was if there is an example to connect a machine ( stm32f4 ) to the host through socketCAN, I mean by the host here is the machine running renode. For the example that you shared it was a communication between two emulated machines ( mr_canhubk3 ) communicating through an emulated CANhub on renode.
A communication between host and an emulated machine will be through the virtual CAN ( vcan ) it's a kernel module on linux machines, but I don't know if there is anyway to forward the canHub packets to the vcan.

Mes-ser, 2024-05-01:
Look at this commit https://github.com/renode/renode/commit/7678bcaf83658184aac9c6545ca327f07b7d14a3
And you can download nightly build from https://builds.renode.io/

And can confirm - It work, I used it.

yassinedevop, 2024-05-08:
I tried it with my stm32f4 which has a classic CAN and it's working.
Thanks for the help !

closed 2024-05-08 by yassinedevop