renode/renode #599: Usage with EFM32GG, board STK3700

SpasicRebeka, 2024-04-17:
### Description

I set up Renode as described, then tried it out with a simple program, which should set a LED.

I used this peripheral config file for it: [stk3700.zip](https://github.com/renode/renode/files/15017631/stk3700.zip).

I uploaded its binary to my physical board and it works properly, I'm gonna attach its source here: [renode_test.zip](https://github.com/renode/renode/files/15017610/renode_test.zip)

The binary, I've wanted to run with renode is located here: renode_test\GNU ARM v7.2.1 - Debug\renode_test.bin

Whenever I try to read the status of the LED in question, it returns false.

### Expected behaviour

The status of the LED should be true.

### How to reproduce?

I've already described the method above how I got here.

Some bonus information:

the commands I sent to renode:

----  
Renode, version 1.14.0.34213 (81da46c1-202308081855)
(monitor) mach create "test"
(test) machine LoadPlatformDescription @platforms/boards/silabs/stk3700.repl 
(test) peripherals
 Available peripherals:
  sysbus (SystemBus)
  │   
  ├── bitband (BitBanding)
  │       <0x42000000, 0x43FFFFFF>
  │       
  ├── cmuStatusOscillator (PythonPeripheral)
  │       <0x400C802C, 0x400C802F>
  │       
  ├── cpu (CortexM)
  │       Slot: 0
  │       
  ├── deviceInformation (EFM32xGDeviceInformation)
  │       <0x0FE081B0, 0x0FE081FF>
  │       
  ├── flash (MappedMemory)
  │       <0x00000000, 0x000FFFFF>
  │       
  ├── gpioPort (EFMGPIOPort)
  │   │   <0x40006000, 0x4000613F>
  │   │   
  │   ├── led0 (LED)
  │   │       Address: 66
  │   │       
  │   ├── led1 (LED)
  │   │       Address: 67
  │   │       
  │   ├── button0 (Button)
  │   │       Address: 64
  │   │       
  │   └── button1 (Button)
  │           Address: 65
  │           
  ├── i2c0 (EFM32GGI2CController)
  │       <0x4000A000, 0x4000A3FF>
  │       
  ├── i2c1 (EFM32GGI2CController)
  │       <0x4000A400, 0x4000A7FF>
  │       
  ├── leUart0 (LEUART)
  │       <0x40084000, 0x400843FF>
  │       
  ├── leUart1 (LEUART)
  │       <0x40084400, 0x400847FF>
  │       
  ├── nvic (NVIC)
  │       <0xE000E000, 0xE000EFFF>
  │       
  ├── sram (MappedMemory)
  │       <0x20000000, 0x2001FFFF>
  │       
  ├── timer0 (Efm32Timer)
  │       <0x40010000, 0x400103FF>
  │       
  ├── timer1 (Efm32Timer)
  │       <0x40010400, 0x400107FF>
  │       
  ├── timer2 (Efm32Timer)
  │       <0x40010800, 0x40010BFF>
  │       
  ├── timer3 (Efm32Timer)
  │       <0x40010C00, 0x40010FFF>
  │       
  ├── uart0 (EFM32_UART)
  │       <0x4000E000, 0x4000E3FF>
  │       
  ├── uart1 (EFM32_UART)
  │       <0x4000E400, 0x4000E7FF>
  │       
  ├── usart0 (EFM32_UART)
  │       <0x4000C000, 0x4000C3FF>
  │       
  ├── usart1 (EFM32_UART)
  │       <0x4000C400, 0x4000C7FF>
  │       
  ├── usart2 (EFM32_UART)
  │       <0x4000C800, 0x4000CBFF>
  │       
  └── vcmpStatus (PythonPeripheral)
          <0x40000008, 0x4000000B>
(test) sysbus LoadBinary @C:/Users/rebek/renode_test.bin 0x000
(test) start
Starting emulation...
(test) sysbus.gpioPort.led0 State
False
(test) pause

----  

I turned on the function name logging and got:

----  
...
23:37:47.3880 [WARNING] gpioPort: Unhandled read from offset 0x94.
23:37:47.3880 [WARNING] gpioPort: Unhandled write to offset 0x4, value 0x400.
23:37:47.3880 [INFO] cpu: Entering function without name at 0x6E2
23:37:47.3880 [INFO] cpu: Entering function without name at 0x704
23:37:47.3880 [INFO] cpu: Entering function without name at 0x728
23:37:47.3880 [INFO] cpu: Entering function without name at 0x71C
23:37:47.3880 [INFO] cpu: Entering function without name at 0x5F8
23:37:47.3880 [INFO] cpu: Entering function without name at 0x610
23:37:47.3880 [INFO] cpu: Entering function without name at 0x616
23:37:47.3880 [INFO] cpu: Entering function without name at 0x61C
23:37:47.3880 [INFO] cpu: Entering function without name at 0x622
23:37:47.3880 [INFO] cpu: Entering function without name at 0x62C
23:37:47.3880 [INFO] cpu: Entering function without name at 0x632
23:37:47.3880 [INFO] cpu: Entering function without name at 0x638
23:37:47.3880 [INFO] cpu: Entering function without name at 0x5A0
23:37:47.3880 [INFO] cpu: Entering function without name at 0x5B2
23:37:47.3880 [INFO] cpu: Entering function without name at 0x5B8
23:37:47.3880 [INFO] cpu: Entering function without name at 0x5BE
23:37:47.3880 [INFO] cpu: Entering function without name at 0x5C4
23:37:47.3880 [INFO] cpu: Entering function without name at 0x5CE
23:37:47.3880 [INFO] cpu: Entering function without name at 0x642
23:37:47.3880 [INFO] cpu: Entering function without name at 0x64E
23:37:47.3880 [INFO] cpu: Entering function without name at 0x654
23:37:47.3880 [WARNING] gpioPort: Unhandled read from offset 0x94.
23:37:47.3880 [WARNING] gpioPort: Unhandled write to offset 0x4, value 0x400.
23:37:47.3880 [INFO] cpu: Entering function without name at 0x6E2
23:37:47.3880 [INFO] cpu: Entering function without name at 0x704
23:37:47.3880 [INFO] cpu: Entering function without name at 0x728
23:37:47.3880 [INFO] cpu: Entering function without name at 0x71C
23:37:47.3880 [INFO] cpu: Entering function without name at 0x5F8
23:37:47.3880 [INFO] cpu: Entering function without name at 0x610
23:37:47.3880 [INFO] cpu: Entering function without name at 0x616
23:37:47.3880 [INFO] cpu: Entering function without name at 0x61C
23:37:47.3880 [INFO] cpu: Entering function without name at 0x622
23:37:47.3880 [INFO] cpu: Entering function without name at 0x62C
23:37:47.3880 [INFO] cpu: Entering function without name at 0x632
23:37:47.3880 [INFO] cpu: Entering function without name at 0x638
23:37:47.3880 [INFO] cpu: Entering function without name at 0x5A0
23:37:47.3880 [INFO] cpu: Entering function without name at 0x5B2
23:37:47.3880 [INFO] cpu: Entering function without name at 0x5B8
23:37:47.3880 [INFO] cpu: Entering function without name at 0x5BE
23:37:47.3880 [INFO] cpu: Entering function without name at 0x5C4
23:37:47.3880 [INFO] cpu: Entering function without name at 0x5CE
23:37:47.3880 [INFO] cpu: Entering function without name at 0x642
23:37:47.3880 [INFO] cpu: Entering function without name at 0x64E
23:37:47.3880 [INFO] cpu: Entering function without name at 0x654
...

----  

### Environment

- OS: Win11
- Renode version (with commit SHA):  version 1.14.0.34213 (81da46c1-202308081855)

### Additional information

I tried to debug also with gdb, without any success. The binary file was said to be not in executable format: File format not recognized

### Do you plan to address this issue and file a PR?

No, I just need help.

JayJay-K, 2024-04-18:
Hi,
I don't have deep knowledge about Renode. But I think we can refer "https://github.com/renode/renode/issues/352"

We can try to add the following to connect between GPIO pin to LED:

gpioPort:
    66 -> led0@0
    67 -> led1@0

I'm not sure it's a proper answer. But I had same issue before.
I think we can try it :)

SpasicRebeka, 2024-04-22:
Appreciate your help, but unfortunately didn't solve my problem... :(((

tobermory, 2024-05-30:
I am familiar with the STK3700 and Silabs EFM32 cpus in general, less so with Renode. I found other applications for EFM32 where Renode would not run code as real hardware would.
This line, from your notes
```
23:37:47.3880 [WARNING] gpioPort: Unhandled read from offset 0x94.
```
is I think a read from a GPIO_P_Typedef.MODEL register. Refer to Silabs headers efm32gg_gpio.h and efm32gg_gpio_p.h for instance. I think you code line to an emlib call
```
GPIO_PinModeSet(gpioPortE,2, gpioModePushPull, 1);
```
will access the MODEL register for the E port (hence offset 0x94).

I located a file EFMGPIOPort.cs in the Renode source code tree. It is referenced by the efm32gg242.repl file I think you are using:
```
gpioPort: GPIOPort.EFMGPIOPort @ sysbus 0x40006000
```

I think the WriteDoubleWord impl shows that the MODEL register is basically not implemented ??

Hope this helps, and sorry it's not good news.