renode/renode #602: STM32_PWR peripheral initialization

miguelgarcia-ITI, 2024-04-19:
### Description

This issue is related to the _STM32_PWR_ peripheral, defined in _Miscellaneous/STM32_PWR.cs_.

A real **STM32F4xx Board** initializes the **PWR peripheral** in a different manner than Renode does. Specifically, Register _PWR_CSR_ (PowerControlStatus at offset 0x04) is expected to initialize the _VOSRDY_ Flag (bit 14) by default as 1.

When using STM Cube IDE to configure a project and automatically generate the source code of your application, the generated initialization process, in the function _SystemClock_Config()_, waits until _PWR.VOSRDY_ flag is equal to 1. And this never happens.

### Questions

Is there a method to set the value of this flag using a command in the .resc file? avoiding modifying the application source code. It is essential for us to test the same code in Renode and in the stm32 board. 

Or is the only way to do that to modify the implementation of the _STM32_PWR_ peripheral? Changing the definition of the _VOSRDY_ flag from _WithTaggedFlag_ to _WithFlag_.

miguelgarcia-ITI, 2024-04-22:
This issue is associated with PR#103 in renode-infrastructure

pgielda, 2024-04-25:
https://github.com/renode/renode-infrastructure/pull/103
FYI @PiotrZierhoffer @mateusz-holenko

mateusz-holenko, 2024-04-30:
Thanks @miguelgarcia-ITI for reporting the problem and creating the PR.
Your proposed fix is now merged upstream.

closed 2024-05-07 by miguelgarcia-ITI