renode/renode #603: Way to jump to specific PC while running the firmware directly via monitor or pyrenode?

asmitaj08, 2024-04-25:
Hi,
In renode, is there way to jump to specific PC while running the firmware directly via monitor or pyrenode without using GDB?

MateuszKarlic, 2024-04-25:
Hi @asmitaj08 

You can change PC by executing `cpu PC new_address` in Monitor. However, please note that doing it on running simulation might lead to unexpected results, especially if handling interrupts is involved. You should attempt to pause the emulation first by typing `pause`.

Also, please note that this will only change the Program Counter. No other registers will be modified, and no return address will be saved. This might impact how your program behaves.

If you need more flow-control features in Monitor, please take a look at: https://renode.readthedocs.io/en/latest/basic/control.html or Python scripting utility: https://renode.readthedocs.io/en/latest/basic/using-python.html#cpu-hooks

asmitaj08, 2024-04-25:
Thank you!
Yeah, I completely forgot about CPU hooks.
I basically wanted to restore register states (PC, SP, LR), and also wanted to write custom values at specific memory addresses. 
I know of Save and Load feature, but was wondering if I could do it by writing at each of these registers, and addresses.

asmitaj08, 2024-04-25:
Thanks a million!

asmitaj08, 2024-04-26:
Hi,
It works fine with .resc. But I am now trying to do it using python (took help from pyrenode3).
Most of the things like get address via symbol name `mach.sysbus.GetSymbolAddress("main")`, getting and setting registers , `mach.sysbus.cpu.GetRegisterUnsafe(13).RawValue` works, but when I similarly do `AddHook` as : `mach.sysbus.cpu.AddHook(main_func_addr,"print('hello')")`, it gives me type error as :

```
mach.sysbus.cpu.AddHook(main_func_addr,"print('hello')")
TypeError: No method matches given arguments for TranslationCPU.AddHook: (<class 'int'>, <class 'str'>)
```
But the AddHook method parameters are correct as per my knowledge,` "void AddHook(uInt64 addr, String pythonScript)"`

Do you have any suggestions?
Any insight would be really helpful.
Thank you!!

MateuszKarlic, 2024-04-26:
You could try to use
```python3
import System
from Antmicro.Renode.Peripherals.CPU import ICpuSupportingGdb

def hook(cpu, address):
    print(address)

Action = getattr(System, 'Action`2')
hook_action = Action[ICpuSupportingGdb, System.UInt64](hook)
mach.sysbus.cpu.AddHook(0xDEADBEEF, hook_action)
```
Note that this does not place you within the internal Python runtime running in Renode, but on the pyrenode's side. This might be even more useful for you, since you will have access to all of pyrenode's API.

asmitaj08, 2024-04-28:
Thank you so much. Yes, this works. I am kind of unable to understand, `'Action`2'` , what does that exactly mean, where is it defined (looks like system object, sorry I don't have clear understanding of c#). Is it gonna be 'Action`2' in all cases , like even if I do it for `SetHookAtBlockBegin` ?

MateuszKarlic, 2024-04-29:
I'm no expert on Pythonnet bindings myself, but it pulls a type `Action` - action being generic, supports two generic parameters here - hence ``Action`2`` - see: https://learn.microsoft.com/en-us/dotnet/api/system.action-2. It's a type defined in C# "runtime".
In the next line we create a concrete instance of this type - `hook_action = Action[ICpuSupportingGdb, System.UInt64](hook)` - with these template parameters in brackets, and we pass the function (`hook`) to this `Action` instance.

Action is a `delegate` - you can think of it like a pointer to a function. These two template arguments, are the arguments of the function that it intends to point to. So in this case the target function will be of this format: `void hook_function(ICpuSupportingGdb cpu, System.UInt64 address)`. And then in parenthesis, you pass the function name itself.

So to answer your question - you will have to instantiate your `Action` type differently depending on the hook - with different type parameters in the brackets. Or if a hook requires more parameters, then you will have to use ``Action`3``, etc. 

For `SetHookAtBlockBegin` you can still use ``Action`2``, but with a bit different generic params, likely something like this:
```python3
Action = getattr(System, 'Action`2')
hook_action = Action[System.UInt64, System.UInt32](hook)
```

asmitaj08, 2024-04-29:
Got it. Thank you so much.

asmitaj08, 2024-05-08:
Hi,
I want to check if it is possible to pauses a machine inside hook_function, and resume it externally in main function? (I don't think it's possible, but was just wondering if there is any way). Right now I am doind it using a flag, where hook sets the flag, and then I do required tasks in the main function dependeing on the flag value, but that not accurate.  Is there any other suggestion? Thank you!

MateuszKarlic, 2024-05-10:
As long as you only have one CPU running in the simulation, this actions should work without problems. CPU is not executing instructions when inside a hook.

As for pausing: I'm not sure I understand your scenario fully, but if you just pause machine inside the `hook_function` it will be precise, exactly at the position (address) where the hook is registered. You can then resume the machine from anywhere in your code you want.

asmitaj08, 2024-05-11:
Got it, thanks. Yeah, it works sometimes. Sometimes, just doesn't respond, especially, when I do SaveState inside  the hook, using `m.execute(f'Save @{state_file}')`

asmitaj08, 2024-05-24:
Hi,
A gentle follow up, wanted to check if doing `m.execute(f'Save @{state_file}')` is a good idea? Or is there any alternative. My goal is to kinda get a snapshot at a particular fucntion, and reload from that stage again instead of from start. Earlier, I tried doing by just saving LR,SP,PC, but I guess if I save all the contexts (using renode Save feature), it should be better.

MateuszKarlic, 2024-05-24:
Hi, depending what is your actual goal, I think this can be a good solution - though this saves the state of the entire emulation, including all peripherals, and memory contents, not only of the CPU.  There is currently no way to preserve the state selectively of only one peripheral (or just of the one CPU), but this is not something that you would usually want. See here: https://renode.readthedocs.io/en/latest/basic/saving.html for description of this feature

If you experience issues specifically with this function, you could create a separate issue, with a reproduction template