renode/renode #607: Add command line display of `renode` version?

g40, 2024-04-29:
Using a fresh install of `renode_1.9.0.msi` fails to run as expected. `

`--version` is unsupported and
`--help`  displays no salient version information.

Perhaps it should?

```
>renode --version
Unexpected options detected: --version

usage: Renode.exe [options] [script]
```

Mes-ser, 2024-04-30:
I would recommend updating to newest version (1.15.0) where `--version`/`-v` is working.

g40, 2024-04-30:
@PiotrZierhoffer @Mes-ser 

Thank you for the comments. Here, then, is a bit of a puzzle. I downloaded an installer via Github releases last night as you can see. Please note the version number! 

I will get the latest shortly.

BR 

Jerry.

[image]

mgielda, 2024-05-07:
I think this can be safely closed, if I am misunderstanding anything, please reopen.

closed 2024-05-07 by mgielda