renode/renode #612: error when running HSS in Renode with PolarFire SoC Icicle Example .resc files

sarah18-z, 2024-05-14:
### Description

I'm just getting started with the PolarFire SoC, and I'm trying to run a custom build of the HSS in Renode. Earlier this week, I was able to run the default **icicle-kit.resc** script from Renode without any issues. This script launches HSS, U-Boot, and Linux.

The next step is to modify it to use only HSS and boot with FreeRTOS.
First, I tried to configure the HSS, build it, and add it to the script to run it on my emulated machine. However, I encountered an error during the DDR training process.

**issue :**
[image]

[image]

### Expected behaviour

The machine starts emulating with the customized HSS.

### How to reproduce?
Just configure your HSS using SoftConsole, then build it, and modify the Icicle Kit Renode script as follows.

> My renode Scripts :
```
:description: This is a sample script running HSS, U-Boot and Linux on Icicle Kit with PolarFire SoC

$bin?=@/home/sarah-io/Desktop/work-io/testsar/hss-envm-wrapper.elf
$uboot?=@https://dl.antmicro.com/projects/renode/icicle--u-boot-s_5132448-194bf14572a9bc4b26727567065ede2ffd7f1201
$vmlinux?=@https://dl.antmicro.com/projects/renode/icicle--vmlinux-s_8563992-fa2aad1e61ec38b411f6afb543503cb26601b1e2
$fit?=@https://dl.antmicro.com/projects/renode/icicle--fitImage.fit-s_16976563-1d0e86ed4cc7c24e167ca899fd929d954956b478
$emmc?=@https://dl.antmicro.com/projects/renode/icicle--emmc.img-s_26746880-3a6ef871bc8eb6fcfbda344e8c23fb534ef89961

i @scripts/single-node/polarfire-soc.resc

# these settings guarantee a deterministic, successful boot-up of Linux
emulation SetGlobalSerialExecution true
emulation SetGlobalQuantum "0.0008"

showAnalyzer mmuart1
machine SdCardFromFile $emmc mmc 0x100000000 

macro reset
"""
    sysbus LoadBinary $fit 0x88300000
    sysbus LoadSymbolsFrom $vmlinux
    sysbus LoadSymbolsFrom $uboot
"""
runMacro $reset
```
Then launch your emulation in Renode.

### Environment

Please, provide the following information:

- OS : Linux `Ubuntu 20.04`
- Renode version :  `Renode v1.15.0.27975`  `build: 9111b18e-202403181532` `build type: Release` `runtime: Mono 4.0.30319.42000`

### Additional information

nothing to add. 

### Do you plan to address this issue and file a PR?

No, I just need help.

mateusz-holenko, 2024-05-24:
Hi @sarah18-z, thanks for reporting the issue.

DDR controller for Icicle Kit is currently mocked and might not be versatile enough to support a newer version of HSS.

That being said, from the Renode perspective, there is no need to actually train DDR memory before use (we simulate it on a slighty higher abstraction level).
If this procedure is not crucial from the perspective of your application you could try using the following hook to skip execution of the problematic fragment:
```
e51 AddHook `sysbus GetSymbolAddress "HSS_DDRInit"`
"""
value = 1
cpu.SetRegisterUnsafe(10, Antmicro.Renode.Peripherals.CPU.RegisterValue.Create(value, 32))
cpu.PC = cpu.RA
"""
```
After adding this hook to your resc script, every time software calls the `HSS_DDRInit` function CPU will automatically set the return code (register 10) and jump back as if the function executed to the return statement.

This should allow you to pass the initialization code without the need of changing your binary.

sarah18-z, 2024-05-26:
Hello @mateusz-holenko , thank you for your response and for your interest in my issue. I modified the resc script by adding the hook that you suggested, but I still encounter the same error.

sarah18-z, 2024-05-26:
Hello, @mateusz-holenko thank you for your response and for your interest in my issue. 
I modified the resc script by adding the hook that you suggested, but I still encounter the same error.
[image]

mateusz-holenko, 2024-05-27:
The error in Monitor indicates that there is no `HSS_DDRInit` symbol in the file you load (so Renode can't attach to it to skip execution).
How do you build the HSS image, could you share the elf file?

sarah18-z, 2024-06-24:
@mateusz-holenko  I have some updates: I managed to resolve the DDR insertion error by using HSS version 0.99.33, and it now works correctly with Renode version 1.15.0. To address DDR training, I disabled it in my HSS configuration.

However, a new issue has surfaced: the HSS is unable to boot from the eMMC, and I'm unsure how to solve this problem.

[image]