renode/renode #624: STM32H7 SPI

RallySmith, 2024-05-22:
### Description

The current renode STM32H7 models do not implement complete SPI model functionality.

See issue template mirror:
https://github.com/RallySmith/renode-issue-stm32h7-spi

NOTE: Not a fork since github limits users to a single fork and I have a fork open for the pending issue:
#606 and so I just mirrored the repo and am referencing the new repo from this issue report; but it means it does not appear at: https://github.com/renode/renode-issue-reproduction-template/forks

### Expected behaviour

The `fixed` branch supplied STM32H7_SPI_Fixed.cs model provides updates to:

- fix typo in DMAReceive naming
- add DMATransmit and transmitDMAEnabled
- avoid always calling peripheral TransmissionFinished() at the end of a transfer
- track "unhandled" Configuration2 fields to avoid Warnings and return state if read by device driver

### How to reproduce?

The `main` branch will fail to complete after triggering a warning about the unhandled PWR controller. The `fixed` branch will successfully execute and complete the SPI operations.

### Additional information

The `fixed` branch STM32H7_PWR.cs source file is identical to the version in #618 and #623

The `fixed` branch STM32DMA_Fixed.cs and STM32H7_DMAMUX.cs sources are identical to the versions in #618

### Do you plan to address this issue and file a PR?

Yes, at a point when the maintainers are happy with the fixes as provided in the fixed branch in the mirror.

DenSandu, 2026-02-13:
Hi, do I need to download only `STM32H7_SPI_Fixed.cs` file in order to solve the spi communication? Or do I also need 
- [STM32DMA_Fixed.cs‎](https://github.com/RallySmith/renode-issue-stm32h7-spi/commit/f17f43b2d156855130d90fb7ad9153a7c0515de9#diff-4feaea5764756303001e16dc0bf29894406001a05c980260f2cbb191ec73f6c0)
- [STM32H7_DMAMUX.cs](https://github.com/RallySmith/renode-issue-stm32h7-spi/commit/f17f43b2d156855130d90fb7ad9153a7c0515de9#diff-c95cc8f3070fdd80f3f7f65a491355426682306f6d9969b7f20231c6ed10c86e)
- [STM32H7_PWR.cs](https://github.com/RallySmith/renode-issue-stm32h7-spi/commit/f17f43b2d156855130d90fb7ad9153a7c0515de9#diff-6343adff0400c262fd8d4a76484e029c72253730485bb41337c4fdd67f6fa04c)
- [SPI_Fixed.cs‎](https://github.com/RallySmith/renode-issue-stm32h7-spi/commit/f17f43b2d156855130d90fb7ad9153a7c0515de9#diff-e6c08d1bdd6e5cd292fd2ba418738e58c7be827391102e9171d5038c0ce6f167)

RallySmith, 2026-02-13:
If you intend to run software that makes use of SPI DMA transfers then 100% you will need the DMA fixes.