renode/renode #625: STM32_Timer Produces error

emtee14, 2024-05-22:
### Description

Renode throws the error below when implementing the STM32_Timer in the platform file. I've removed all other peripherals and the issue still happens. It works without the timer though.

```bash
There was an error executing command 'machine LoadPlatformDescription @C:\Users\morgan\CLionProjects\hamilton-firmware\renode\STM32L4.repl'
Error E21: Could not find any suitable constructor for this registration point.
At C:\Users\morgan\CLionProjects\hamilton-firmware\renode\STM32L4.repl:132:37:
timer2: Timers.STM32_Timer @ sysbus 0x40000000
```

### Expected behaviour

It should load the platform and then run the firmware

### How to reproduce?

All the files are in the repo https://github.com/emtee14/renode-issue-reproduction-template/tree/main I run it with the command ```renode --console -e s HFC_v1.resc```. You shouldn't need the firmware as it happens before it's loaded.
### Environment

Please, provide the following information:

- OS (macOS and windows)
- Renode version 
Renode v1.15.0.30170
  build: 9111b18e-202403181638
  build type: Release
  runtime: .NET Framework 4.0.30319.42000

### Do you plan to address this issue and file a PR?
I will need support with the initial diagnosis but will be happy to write the fix if it is a bug.

Thanks for the help

mateusz-holenko, 2024-05-23:
Hi @emtee14, thanks for reporting the issue.

`STM32_Timer` requires two paramters: `frequency` and `initialLimit` (see https://github.com/renode/renode-infrastructure/blob/master/src/Emulator/Peripherals/Peripherals/Timers/STM32_Timer.cs#L24 for details).

As a reference on how to set them, take a look at: https://github.com/renode/renode/blob/master/platforms/cpus/stm32f0.repl#L123-L126

emtee14, 2024-05-23:
Thanks for your reply @mateusz-holenko. I have the parameters set correctly (fairly sure) and that didn't work. I copied and pasted the implementation from the stm32f0, which still gave me the same error. 
https://github.com/emtee14/renode-issue-reproduction-template/blob/d601660bc63af5cacb0f60dee20a776bee08504a/renode/STM32L4.repl#L132C1-L135C29

MerinoSheep, 2024-05-30:
@emtee14 I was having a similar issue on Win11 trying to load a local platform and using a nightly build has temporarily solved it.

emtee14, 2024-05-30:
Thanks @MerinoSheep I'll give that a go

MateuszKarlic, 2024-05-31:
I believe, that your issues might come from using older version of Renode, with newer version of the platform description (modified after release `1.15.0`).  I checked with this release and was able to reproduce the error.

In https://github.com/renode/renode-infrastructure/commit/f64afc98fa8a7e2d3135bbe4f573583947c4eba7 we changed the way `STM32_Timer` can be declared in the REPL - you no longer need to specify size as was the case before these changes were made: https://github.com/renode/renode/blob/3c29e1251bb5d68651e7592255cd86432da5a852/platforms/cpus/stm32f0.repl#L123 

These changes aren't present in `1.15.0` release, but you attempt to use a REPL that assumes that these changes have been already introduced.

So the solution is indeed, to either use new Renode, built **after** these changes are introduced (in commit https://github.com/renode/renode/commit/cb836ef05c87f9b4dd60c2c27d4b9989809c1ff1), or modify your platforms, as shown in the link above, to include the size of the peripheral.

emtee14, 2024-06-03:
Yeah that works thank you

closed 2024-06-03 by emtee14