renode/renode #636: `InvalidOperationException` is thrown when invoking a method via reflection

otavio, 2024-06-20:
### Description

An `InvalidOperationException` is thrown when invoking a method via reflection within Renode 1.15.1. The exception occurs specifically at the point of invoking the method using `System.Reflection.MonoMethod.Invoke`.

### Expected behaviour

The method should be invoked successfully without any exceptions, provided that the method parameters and state are valid. However, an `InvalidOperationException` is thrown during the method invocation.

### How to reproduce?

Is there a recommended way to gather more detailed information about the internal state of the method or the Renode environment that could help in diagnosing this issue? Any advice or tools available within Renode for this purpose would be greatly appreciated.

### Environment

- OS: Linux 6.9.4
- Renode v1.15.1.23066
  build: bef0c7b3-202406141248
  build type: Release
  runtime: Mono 4.0.30319.42000

### Do you plan to address this issue and file a PR?

Sure, but first, I need to understand the cause of the error.

Cc: @nandojve

otavio, 2024-06-20:
Log:

```
InvalidOperationException:   at (wrapper managed-to-native) System.Reflection.MonoMethod.InternalInvoke(System.Reflection.MonoMethod,object,object[],System.Exception&)
  at System.Reflection.MonoMethod.Invoke (System.Object obj, System.Reflection.BindingFlags invokeAttr, System.Reflection.Binder binder, System.Object[] parameters, System.Globalization.CultureInfo culture) [0x0006a] in <aeea2e33ea654cfabae63b60ba0e127a>:0 
```

mateusz-holenko, 2024-07-04:
Hi @otavio, could you provide a short reproducible example that we could run on our end?

otavio, 2024-08-03:
Using the latest release and .NET build has improved the logs. See below:

```
InvalidOperationException:    at Antmicro.Renode.RobotFramework.UartKeywords.OperationFail(TerminalTester tester)
   at Antmicro.Renode.RobotFramework.UartKeywords.WaitForLineOnUart(String content, Nullable`1 timeout, Nullable`1 testerId, Boolean treatAsRegex, Boolean includeUnfinishedLine, Nullable`1 pauseEmulation, Nullable`1 matchNextLine)
   at InvokeStub_UartKeywords.WaitForLineOnUart(Object, Span`1)
   at System.Reflection.MethodBaseInvoker.InvokeWithManyArgs(Object obj, BindingFlags invokeAttr, Binder binder, Object[] parameters, CultureInfo culture)
```

@mariucker and I are working in some way to reproduce the issue in a simple app.

@mateusz-holenko, is there a way to increase renode log or to better find the root 
cause of error?

otavio, 2024-08-06:
We found what was causing the problem, so we fixed the application on our site, and it is running as expected.

closed 2024-08-06 by otavio