renode/renode #637: wish: rtos awareness for GDB server

karlp, 2024-06-21:
Thanks for taking the time to submit a feature request for Renode!

Please familiarize yourself with our [Contributing guidelines](https://github.com/renode/renode/blob/master/CONTRIBUTING.md) first.

### Description

What I'd _love_ is for the renode GDB server to have RTOS aware thread debugging, such that "info threads" and "thread <name> xxxx" type operations worked.  I'm used to this with OpenOCD and friends when working on real hardware, and it feels very raw now not having it :)  I'd personally love FreeRTOS support, but if zephyr gets added first, I could probably go and copy that sufficiently to get freertos support working.

### Usage example
Easier to debug applications using any RTOS with multiple threads.

### Do you plan to address this issue and file a PR?
I have some interest in working on it, but would need some direction how you'd like to see this included.  
At present I would only be up to doing some gross hacking on QueryCommand.cs to try and find some symbols and fill out a thread list for starters.  I have not worked with c# for a long time though, so it would be slow going, even using other debug middleware as references.

PiotrZierhoffer, 2024-07-04:
Hi @karlp ,

Indeed, this is something we have been considering and actually have an internal implementation for Zephyr.

It's not yet in a releasable state, so I don't expect it to be out very soon - but RTOS awareness is definitely a topic we're investigating.

karlp, 2024-07-04:
That works for me :)  You'll get there far faster than I would on this aspect, and I can spend more time on machine/platform things.  I'm happy with closing this "issue" if you've got this sufficiently on your own roadmap.

closed 2024-07-11 by PiotrZierhoffer