renode/renode #640: Bug in Virtqueue implementation

brempusz, 2024-07-07:
### Description

I think there is a bug in Virtqueue implementation:

https://github.com/renode/renode-infrastructure/blob/b83aa9b46e518171f6c061cfe4fd6a5cfa5c734e/src/Emulator/Main/Storage/VirtIO/Virtqueue.cs#L140
in `public void WriteVirtqueueUsed(int chainFirstIndex, bool noInterruptOnUsed)`

The code saves "bytes processed":
`parent.SystemBus.WriteDoubleWord(ringAddress + (ulong)UsedRing.Length, (uint)BytesProcessed);`

But according to VirtIO specification:
https://docs.oasis-open.org/virtio/virtio/v1.2/cs01/virtio-v1.2-cs01.html#x1-540008
2.7.8 The Virtqueue Used Ring
it should be:
"(...)  len the total of bytes written into the buffer."

The same is visible in example code comment:
```
        /* 
         * The number of bytes written into the device writable portion of 
         * the buffer described by the descriptor chain. 
         */ 
        le32 len; 
```

### Expected behaviour

Renode should track and use bytes written instead of bytes processed.

### How to reproduce?

I am implementing virtio devices for renode and I stumbled on an error while working on VirtIOGPIO device.
Part of my code looks like this:
```
  public override bool ProcessChain(Virtqueue vqueue)
        {
            vqueue.ReadDescriptorMetadata();
            if (!vqueue.TryReadFromBuffers(Marshal.SizeOf(typeof(virtio_gpio_request)), out var requestBytes))
            {
                return false;
            }
            if (!Packet.TryDecode<virtio_gpio_request>(requestBytes, out var request))
            {
                return false;
            }

            vqueue.ReadDescriptorMetadata();
            virtio_gpio_response? response = null;
            switch (request.type)
            {
                case VIRTIO_GPIO_MSG.VIRTIO_GPIO_MSG_GET_LINE_NAMES:
                    using (MemoryStream memStream = new MemoryStream(_gpio_names_bytes.Length + 1))
                    {
                        // _gpio_names_bytes is an array pre-populated with names bytes.
                        memStream.WriteByte((byte)VIRTIO_GPIO_STATUS.VIRTIO_GPIO_STATUS_OK);
                        memStream.Write(_gpio_names_bytes, 0, _gpio_names_bytes.Length);
                        var response_bytes = memStream.GetBuffer();
                        if (!vqueue.TryWriteToBuffers(response_bytes))
                        {
                            return false;
                        }

                        // HACK because 
 of https://github.com/torvalds/linux/blob/8a9c6c40432e265600232b864f97d7c675e8be52/drivers/gpio/gpio-virtio.c#L111
                        // BytesProcessed is invalid here because bytes read by TryReadFromBuffers were counted 
                        vqueue.BytesProcessed = response_bytes.Length;
                    }

                    break;

                    // other request types...
            }

            // more code...  
            return true;
        }

```

Without the HACK kernel logs an error from gpio-virtio.c:
https://github.com/torvalds/linux/blob/661e504db04c6b7278737ee3a9116738536b4ed4/drivers/gpio/gpio-virtio.c#L111

### Environment

Please, provide the following information:

WSL on Windows 10
Renode 1.15.1

### Do you plan to address this issue and file a PR?

I tried to open the project in Visual Studio 2022 and Rider 2024.1.4 but I could not build it
I can't test the change so I will not create a PR.

PiotrZierhoffer, 2024-07-09:
Hi @brempusz , thanks for the report.

Is there a way for you to provide an easy reproduction case for this issue? It would be very helpful in moving this topic forward.

brempusz, 2024-07-09:
In this archive [virtio_gpio_bug.tar.gz](https://github.com/user-attachments/files/16146820/virtio_gpio_bug.tar.gz) can find all files needed to run test renode vm (kernel, sbi, devicetree, rootfs, repl/resc files and custom device files: VirtIOEntropyDevice.cs and VirtIOGPIODevice.cs - note: both are WiP).

I compiled linux for renode machine using my buildroot config (https://github.com/brempusz/buildroot_renode) and added some user-space tools for gpio later.

If you start the machine (by default my hack is enabled and BytesProcessed will be overwritten) and inspect dmesg you will   find something like this:
```
[    0.111078] gpio_stub_drv gpiochip0: (virtio0): added GPIO chardev (254:0)
[    0.111112] gpio_stub_drv gpiochip0: registered GPIOs 512 to 514 on virtio0
```

And after logging in (root, no password) you will see this:

```
# lsgpio 
GPIO chip: gpiochip0, "virtio0", 3 GPIO lines
 line  0: "VIRTIO_GPIO_0" unused [input]
 line  1: "VIRTIO_GPIO_1" unused [input]
 line  2: "VIRTIO_GPIO_2" unused [input]
```

If you modify VirtIOEntropyDevice.cs and comment out lines 119 and 199 (`vqueue.BytesProcessed = responseBytes.Length;`) dmesg will contain:

```
[    0.110587] gpio_virtio virtio0: GPIO operation returned incorrect len (43 : 51)
[    0.110690] gpio_virtio virtio0: Failed to get GPIO names: -22
[    0.110810] gpio_virtio virtio0: GPIO operation returned incorrect len (2 : 10)
[    0.110924] gpio_virtio virtio0: GPIO operation returned incorrect len (2 : 10)
[    0.111039] gpio_virtio virtio0: GPIO operation returned incorrect len (2 : 10)
[    0.111558] gpio_stub_drv gpiochip0: (virtio0): added GPIO chardev (254:0)
[    0.111591] gpio_stub_drv gpiochip0: registered GPIOs 512 to 514 on virtio0
```

Difference in the length is 8: this is because it also counts in request (which is 8 bytes long:  ushort type, ushort gpio and uint value => 2 + 2 + 4).

Also, output of `lsgpio` will be different:
```
# lsgpio 
GPIO chip: gpiochip0, "virtio0", 3 GPIO lines
 line  0: unnamed unused [input]
 line  1: unnamed unused [input]
 line  2: unnamed unused [input]
```

Hope this helps.

PiotrZierhoffer, 2024-07-11:
Hi @brempusz ,

Can you please try with the latest version? Current HEAD contains changes that should help in your case

brempusz, 2024-07-11:
Hi, 

It works as expected now. Thanks!
When (if?) I finish the implementation I will 

Tested without my hack:
```
 dmesg | grep -i virtio
[    0.000000] Machine model: riscv-virtio,qemu
[    0.106924] virtio-mmio 10008000.virtio_mmio: Wrong magic value 0x00000000!
[    0.107154] virtio-mmio 10007000.virtio_mmio: Wrong magic value 0x00000000!
[    0.107350] virtio-mmio 10006000.virtio_mmio: Wrong magic value 0x00000000!
[    0.107545] virtio-mmio 10005000.virtio_mmio: Wrong magic value 0x00000000!
[    0.107742] virtio-mmio 10004000.virtio_mmio: Wrong magic value 0x00000000!
[    0.111071] gpio_stub_drv gpiochip0: (virtio0): added GPIO chardev (254:0)
[    0.111105] gpio_stub_drv gpiochip0: registered GPIOs 512 to 514 on virtio0
[    0.298725] virtio_blk virtio2: 1/0/0 default/read/poll queues
[    0.305294] virtio_blk virtio2: [vda] 122880 512-byte logical blocks (62.9 MB/60.0 MiB)
# lsgpio 
GPIO chip: gpiochip0, "virtio0", 3 GPIO lines
 line  0: "VIRTIO_GPIO_0" unused [input]
 line  1: "VIRTIO_GPIO_1" unused [input]
 line  2: "VIRTIO_GPIO_2" unused [input]
```

One more (minor) thing: I noticed that paths/namespaces are messed up for virtio stuff:

|path|namespace|class|
|---|---|---|
|`src/Emulator/Main/Storage/VirtIO/VirtIO.cs` | `Antmicro.Renode.Storage.VirtIO`|`VirtIO`|
|`src/Emulator/Main/Storage/VirtIO/VirtIOMMIO.cs` | `Antmicro.Renode.Peripherals.Storage`|`VirtIOMMIO`|
|`src/Emulator/Main/Storage/VirtIO/Virtqueue.cs` | `Antmicro.Renode.Storage.VirtIO`|`Virtqueue`|

I think it should be something like this:

|path|namespace|class|
|---|---|---|
|`src/Emulator/Main/Peripherals/VirtIO/VirtIO.cs` | `Antmicro.Renode.Peripherals.VirtIO`|`VirtIO`|
|`src/Emulator/Main/Peripherals/VirtIO/VirtIOMMIO.cs` | `Antmicro.Renode.Peripherals.VirtIO`|`VirtIOMMIO`|
|`src/Emulator/Main/Peripherals/VirtIO/Virtqueue.cs` | `Antmicro.Renode.Peripherals.VirtIO`|`Virtqueue`|