renode/renode #641: CAN tools (Wireshark, candump, python-can) run directly from host shows invalid CAN ID while using SocketCAN feature

Mes-ser, 2024-07-09:
### Description

While working with SocketCAN feature I stumble across issue, where my CAN Frame ID is byte-reversed while using tools like `candump` or `wireshark` (Run from host directly - more on that later)
This doesn't occur while running WireShark via:
`emulation LogCANTraffic`

### Expected behaviour

When using external tools (candump / wireshark / python-can module) the byte order is kept intact.

### How to reproduce?
- Create 'vcan0' interface
`> ip link add dev vcan0 type vcan`
`> ip link set up vcan0`
- Run renode and start:
`@scripts/complex/socketcan_bridge/nucleo_h743zi-socketcanbridge.resc`
- Open another terminal window and run candump
`candump vcan0`
- Observe that received ID isn't '1' but '0'

Using Wireshark

### Environment

- ubuntu 24.04
- Official 1.15.1 release

### Additional information

While digging what can cause this behavior in my CAN implementation I started looking at Renode source code, and came across this line
[SocketCANBridge.cs:86](https://github.com/renode/renode-infrastructure/blob/9b87dd6ac7aa5cf93d75c751c7feaa4526c96e11/src/Emulator/Peripherals/Peripherals/CAN/SocketCANBridge.cs#L86)
I changed call parameter from `true` to `false` and it worked, even when run `LogCANTraffic` which I assumed was feature which required that this method is called with `true` value, it seems that it doesn't.

Tested different configurations, changed my CAN peripheral implementation and nothing worked, so checked with official example and its sill doesn't work.
Maybe I'm doing something wrong if so please tell me how I can fix that.

### Do you plan to address this issue and file a PR?
Probably as we need this feature to be available in Release version.

jr1221, 2025-02-22:
We managed to mitigate this by extracting the CAN ID and swapping the bytes manually.  If you need a patch file please let me know.  We would make a PR but this breaks extended CAN IDs.

Mes-ser, 2025-02-24:
Thanks but I have patch for that. simply changing
`frame = message.ToSocketCAN(true);` to `frame = message.ToSocketCAN(false);` in [/src/Emulator/Peripherals/Peripherals/CAN/SocketCANBridge.cs:86](https://github.com/renode/renode-infrastructure/blob/9b87dd6ac7aa5cf93d75c751c7feaa4526c96e11/src/Emulator/Peripherals/Peripherals/CAN/SocketCANBridge.cs#L86) and recompiling Renode but downside is that you need to do it by yourself every new release.
This works with Renode examples and my CAN implementation.
I would like to know if this is bug or it's expected behaviour and I'm doing something wrong.

rg-silva, 2025-03-11:
Just hit this issue and looking through the source code I found some seemingly wrong details on the SocketCAN frame conversion, like the endianess, but also, it seems 29 bit ids are used in both Classic and Extended CAN frames, which is wrong... so I think this is a bug.

I'm up to fix this as it affects our testing workflow, but I'd like to know what the reason for the byte swap was initially, so I don't break anything.

starlord-sun, 2025-05-07:
当我打开stm32的can rx中断时（can_it_fmp0）， can设备工作异常，和这个问题相关吗？
When I turn on the can rx interruption of stm32 (can_it_fmp0), the can device works abnormally. Is it related to this problem?

muehlke, 2025-05-15:
I can second what @rg-silva pointed out. There is a bug when it comes to the conversion of frames from and to SocketCAN. I'm working on an STM32F4 using the STMCAN peripheral. In order to convert the message ID properly to a CANMessageFrame that would be interpreted right by the STMCAN peripherals, I had to do some bit shifting to adhere to the format of the RI0R register:

In `CANMessageFrame.TryFromSocketCAN`:

```c#
// adapting the CAN ID accordingly so that when passed
// to CAN peripherals it can be interpreted correctly
// see documentation of register CAN_RIxR
// see constants STIDSHIFT and EXIDSHIFT from Peripherals.CAN.STMCAN.CANMessage
classicalFrame.id = (classicalFrame.id << (classicalFrame.extendedFrameFormat ? 3 : 21));
classicalFrame.id |= (classicalFrame.extendedFrameFormat ? 1u << 2 : 0u);
classicalFrame.id |= (classicalFrame.remoteTransmissionRequest ? 1u << 1 : 0u);
```

On the other side, in order to be able to convert from STMCAN to SocketCAN properly I had to extract two important flags from the CAN message  -which were just set to false - and set them in the constructor of the `CANMessageFrame` in the `TransmitData(CANMessage)` function of the STMCAN peripheral:

```c#
if(FrameSent != null)
{
     // Message frame created with receive identifier register
    // value because filtering can be implemented on bits other
    // than just standard/extended identifier value, e.g. IDE
    // or RTR bit.
    // CHANGE: extracting IDE and RTR flags and passing them to constructor
    bool extendedFormat = (msg.CAN_RIR & (1u << 2)) != 0;
    bool remoteFrame = (msg.CAN_RIR & (1u << 1)) != 0;
    FrameSent(new CANMessageFrame(msg.CAN_RIR, msg.Data, extendedFormat, remoteFrame));
}
```

Having set these flags, the function `ClassicalSocketCANFrame.FromCANMessageFrame` from `SocketCAN.cs` had to be adapted so that it bit shiftes the ID according to the given format:

```c#
 public static ClassicalSocketCANFrame FromCANMessageFrame(CANMessageFrame msg)
        {
            return new ClassicalSocketCANFrame
            {
                // shifting the ID bits according to given format
                id = msg.Id >> (msg.ExtendedFormat ? 3 : 21),
                errorMessageFrame = false,
                remoteTransmissionRequest = msg.RemoteFrame,
                extendedFrameFormat = msg.ExtendedFormat,
                length = msg.Data.Length,
                data = msg.Data.CopyAndResize(MaxDataLength)
            };
        }
```

To my surprise, and the reason why I wrote this answer in this question is that I also had to change `frame = message.ToSocketCAN(true);` to `frame = message.ToSocketCAN(false);` in [/src/Emulator/Peripherals/Peripherals/CAN/SocketCANBridge.cs:86](https://github.com/renode/renode-infrastructure/blob/9b87dd6ac7aa5cf93d75c751c7feaa4526c96e11/src/Emulator/Peripherals/Peripherals/CAN/SocketCANBridge.cs#L86) for the byte order to be correct, otherwise I was getting some funky effects (wrong addresses, remote request frames) on the `candump` of the virtual CAN interface `vcan0`.

This issue raises the question of whether Renode should have some "conversion/translation" layer, so it is able to determine whether a `CANMessage` comes from a certain peripheral (STMCAN, MCAN) and that according to their specific formats (e.g. the format of the STMCAN being the format of the RIxR register) they can be translated properly to a `CANMessageFrame` object, which is (should be) standardized.

Maybe this issue is worthy looking into by the Renode team, specially people who have contributed to SocketCAN, CANHub and SocketCANBridge like @WojMel or @msobkowski

starlord-sun, 2025-06-12:
> After updating yesterday's version, when using socketcan (vcan0) to send a message to renode (stm32f4) with "cansend vcan0 225#112233445566", the data can be received in renode, but both stdid and extid are 0. Is this a bug, or is my configuration incorrect?@WojMel

I also used vcan to bridge the CAN messages of QEMU, but I still can't receive the ID.

qemu==vcan==renode

starlord-sun, 2025-06-27:
Could you confirm this can tx bug?

WojMel, 2025-07-03:
Hi @starlord-sun,

Yes, the fix is already on master.

closed 2025-07-03 by WojMel