renode/renode #642: How to connect multiple AXI masters/slaves to a VerilatedPeripheral

shareefj, 2024-07-10:
Hi there,

I'm starting to use the Zynq-7000 platform with the VerilatorIntegrationLibrary and had a few questions on usage.  We're using both GP0/1 AXI masters and the HP0 slave port in our system but all of the examples in `renode-dpi-examples` only use a single master or slave.  What is the correct path to drive more than 1 AXI interface?

Your `renode.sv` module seems to have parameter support for defining the `BusControllersCount` and `BusPeripheralsCount` but it looks like only values of 0/1 are supported?  I can't see a way to instantiate more than 1 AXI bus and connect it.

Alternatively, are we supposed to be instantiating multiple VerilatedPeripherals in our .repl files?  But then I can't see a way of defining the TCP ports to use.  You can pass the IPv4 address in but not the port numbers so again, it looks like there's only support for a single peripheral at a time.

Thanks.

shareefj, 2024-07-10:
Ah, I see in https://github.com/renode/renode/blob/master/src/Plugins/VerilatorPlugin/Connection/SocketVerilatorConnection.cs#L452 that you don't specify the port and let the OS pick a free port?  And then the user queries the send/receive port using a call to `ConnectionParameters`.

So I think my second option above is the solution - using a different connection for each AXI port?

PiotrZierhoffer, 2024-07-11:
Hi @shareefj ,

Yes, in general, Renode should spawn the verilated process for you here https://github.com/renode/renode/blob/master/src/Plugins/VerilatorPlugin/Connection/SocketVerilatorConnection.cs#L245, whenever you provide the path to the verilated block.

So there is no need to do it manually.

But you are also right that there is no way at the moment to have multiple separate peripherals connected via a single block in the REPL file.

shareefj, 2024-07-11:
Hi @PiotrZierhoffer ,

Thanks but it's the end of a long day and I'm not sure of the relevance of the link to `InitVerilatedProcess`?  Are you saying that's a way of launching the Verilator simulation?  That's not used in your `renode-dpi-examples` is it and it doesn't help for Questa/Xcelium does it?  I think what you were getting at is that's a way of not manually supplying the ports to the simulation if using Verilator?

And can I confirm, if I wanted to connect multiple AXI ports to Renode, they each need to be a separate VerilatedPeripheral?

I've just got my top level compiling with a single AXI port which is fine for now.  Is using the Robot framework the only way of launching a non-Verilator flow or does your Python integration library handle that too?  I'm just about to start looking into that.

Thanks.