renode/renode #643: STM32F103 (Bluepill) simulation with CAN bus

dbarwacz, 2024-07-15:
### Description

I see that [bluepills cpu definitions](https://github.com/renode/renode/blob/master/platforms/cpus/stm32f103.repl) don't include CAN like [STM32F4 does](https://github.com/renode/renode/blob/master/platforms/cpus/stm32f4.repl#L31) (although bluepills support CAN - [reference](https://miro.medium.com/v2/resize:fit:1400/format:webp/0*r7Y5Z7LXcx2Bmy3f.gif))

Is it even possible to tweak the definition to be able to open CAN connector? Thanks!

Mes-ser, 2024-07-16:
in platforms/cpus/stm32f103.repl add
```
can1: CAN.STMCAN @ sysbus <0x40006400, +0x400>
    [0-3] -> nvic@[19-22]

can2: CAN.STMCAN @ sysbus <0x40006800, +0x400>
    [0-3] -> nvic@[63-66]
    master: can1`
```

dbarwacz, 2024-08-07:
Sorry for the delay, I couldn't verify that as I have a problem running simulation with my [simulated chip ](https://github.com/renode/renode/issues/652#issue-2454315404)