renode/renode #646: LTDC add 2 registers (LTDC_SRCR and LTDC_CDSR ) for  stm32h7 support

hattmt, 2024-07-24:
Thanks for taking the time to submit a feature request for Renode!

Please familiarize yourself with our [Contributing guidelines](https://github.com/renode/renode/blob/master/CONTRIBUTING.md) first.

### Description

Need to add 2 registers ( LTDC_SRCR and LTDC_CDSR  for stm32h7 hal support, in the stm32_ltdc peripheral ) 
with bit logic  VBR => VSYNCS

### Usage example

In the HAL on the deinit we have this:

  /* Reload during vertical blanking period */
  __HAL_LTDC_VERTICAL_BLANKING_RELOAD_CONFIG(hltdc);

  /* Get tick */
  tickstart = HAL_GetTick();

  /* Wait for VSYNC Interrupt */
  while (READ_BIT(hltdc->Instance->CDSR, LTDC_CDSR_VSYNCS) == 0U)
  {
    /* Check for the Timeout */
    if ((HAL_GetTick() - tickstart) > LTDC_TIMEOUT_VALUE)
    {
      break;
    }
  }

VSYNCS never pass to one and it's hitting the timeout.

### Additional information

You need to add LTDC_SRCR and  LTDC_CDSR  registers in the [STM32LTDC.cs](https://github.com/renode/renode-infrastructure/blob/master/src/Emulator/Peripherals/Peripherals/Video/STM32LTDC.cs):

   var registerMappings = new Dictionary<long, DoubleWordRegister>
            {
                { (long)Register.BackPorchConfigurationRegister, backPorchConfigurationRegister },
                { (long)Register.ActiveWidthConfigurationRegister, activeWidthConfigurationRegister },
                { (long)Register.BackgroundColorConfigurationRegister, backgroundColorConfigurationRegister },
                { (long)Register.InterruptEnableRegister, interruptEnableRegister },
                { (long)Register.InterruptClearRegister, interruptClearRegister },
                { (long)Register.LineInterruptPositionConfigurationRegister, lineInterruptPositionConfigurationRegister },
                { (long)Register.LTDC_SRCR, new DoubleWordRegister(this)
                    .WithValueField(0, 1, name: "IMR")
                    .WithValueField(1, 1, out var vbr, name: "VBR")
                    .WithReservedBits(2, 30)
                },

                { (long)Register.LTDC_CDSR, new DoubleWordRegister(this)
                    .WithValueField(0, 1, name: "VDES")
                    .WithValueField(1, 1, name: "HDES")
                    .WithValueField(2, 1,FieldMode.Read, valueProviderCallback: _ => vbr.Value, name: "VSYNCS")
                    .WithValueField(3, 1, name: "HSYNCS")
                    .WithReservedBits(4, 28)
                }
            };

here are the registers addresses:
    private enum Register : long
        {
            BackPorchConfigurationRegister = 0x0C,
            ActiveWidthConfigurationRegister = 0x10,
            LTDC_SRCR = 0x024,
            BackgroundColorConfigurationRegister = 0x2C,
            InterruptEnableRegister = 0x34,
            InterruptClearRegister = 0x3C,
            LineInterruptPositionConfigurationRegister = 0x40,
            LTDC_CDSR = 0x48,
        }

### Do you plan to address this issue and file a PR?

Please let us know if you want to work on this yourself or need support.

mateusz-holenko, 2024-08-06:
Hi @hattmat, thanks for reporting the issue.

After looking at the STM32H7 reference manual (https://www.st.com/resource/en/reference_manual/rm0433-stm32h742-stm32h743753-and-stm32h750-value-line-advanced-armbased-32bit-mcus-stmicroelectronics.pdf) we are wondering if the proposed implementation is the optimal way of solving the problem.

Wouldn't just implementing the `LTDC_CDSR` register as returning consecutive values of 0x1, 0x2, 0x4, 0x8, 0x1, 0x2, etc. on each read be enough? The idea would be to imitate a progress in frame display process over time.

Your solution allows simulated software to move forward, but according to the documentation `VSYNCS` and `VBR` flags do not seem to be related to each other; is there any particular reason you decided to implement the logic that way or was it just a configuration that worked for you?

hattmt, 2024-08-07:
Hi @mateusz-holenko ,

Thank you for your reply.

You are right for the logic i used, it was only for my software simulation , too specific. 

So it should be better to implement the real comportement of this register .

I think that what you proposed is the best idea.