renode/renode #648: Dotnet UI not working on Linux

Gerharddc, 2024-07-29:
### Description

When trying to run the "dotnet" version of Renode on Linux, the UI does not launch and I get the following error:

`[WARNING] Couldn't start UI - falling back to console mode`

I see this both when downloading the "linux-portable-dotnet" package from https://github.com/renode/renode/releases and when I try to build and run from source.

### Expected behaviour

I would expect the UI to work on Linux even with .Net 6 since https://renode.io/news/renode-support-for-net-6/ suggests a lot of work was done to make that possible.

I would also expect the error message to provide more information on why the UI failed to start.

### How to reproduce?

1. Download https://github.com/renode/renode/releases/download/v1.15.1/renode-1.15.1.linux-portable-dotnet.tar.gz on a Linux machine
2. Unzip
3. Run the `renode` executable in the extracted folder

### Environment

Please, provide the following information:

- Ubuntu 22.04.4 LTS under WSL 2.2.4.0
- Renode, version 1.15.1.19631 (bef0c7b3-202406141054)

### Additional information

I'm not sure if it makes a difference but I am running on WSL instead of native Linux. This does not seem to be an issue for the Mono packages though as for them the UI launches just fine on WSL.

### Do you plan to address this issue and file a PR?

Unfortunately I doubt I would be able to, but with a lot of help I could try :)

mateusz-holenko, 2024-07-30:
Hi @Gerharddc, thanks for reporting your problem.

Renode UI should work fine on WSL/Linux (both for packages and Renode built from sources), but please remember to install necessary packages: the Mono-based version requires gtk-sharp2 library and .Net-based one gtk-sharp3.

For details, you can take a look at https://github.com/renode/renode-infrastructure/blob/master/src/UI/XwtProvider/XwtProvider.cs#L47-L70.

Gerharddc, 2024-07-30:
@mateusz-holenko thanks, installing `gtk-sharp3` fixed the issue!

I believe it could be very helpful if this were mentioned on https://github.com/renode/renode/blob/master/README.md and https://renode.readthedocs.io/en/latest/advanced/building_from_sources.html since they both currently only mention `gtk-sharp2`.

otavio, 2024-08-03:
I ported the NixOS packages (https://github.com/NixOS/nixpkgs/pull/332064) for the new .NET build and encountered the same issue.

otavio, 2024-08-06:
After changing to the new backend, it worked like a charm. See the change at: https://github.com/NixOS/nixpkgs/pull/332064/files#diff-6e91f99c400b14f01f2e898dfb725181ebd4599c83032c4cdb71a1a815f7a8abR11

Joebeazelman, 2025-08-07:
I'm having the same issue with macOS Tahoe beta.