renode/renode #652: STM32F1 does not finish init code, hanging in `SystemClock_Config`

dbarwacz, 2024-08-07:
### Description
Running an empty project with F103C as a board causes the controller to enter a loop (in this line: https://github.com/stm32duino/Arduino_Core_STM32/blob/5e4f22065273339bbd74c01b8209e9592d2c30bd/variants/STM32F1xx/F103C8T_F103CB(T-U)/variant_PILL_F103Cx.cpp#L109).  
I tested Renode's example config and they run with externally fetched bin file (https://github.com/renode/renode/blob/7a138330ea1d16b064b134e90ebb673d7ce5ef3a/platforms/cpus/stm32f103.repl#L1C1-L80C57 + https://github.com/renode/renode/blob/master/scripts/single-node/stm32f103.resc#L1) 

I suspect default configs shipped with arduino might be affecting how controller initializes (vs how Renode's default Zephyr binary does it) leading to a difference in behaviour/registers that Renode needs to present for the controller's init code to finish the boot successfully.  
What are the troubleshooting steps I can take to figure this one out? 

BTW I noticed I have 
```
No such command or device: init
��Starting emulation...
``` 
right before starting the emulation, I suspect it's from [this](https://github.com/renode/renode/blob/7a138330ea1d16b064b134e90ebb673d7ce5ef3a/platforms/cpus/stm32f103.repl#L78), not sure this might be relevant.

4silvertooth, 2025-02-04:
Did anyone find a solution I am having the exact same issue.

dbarwacz, 2025-02-04:
I unfortunately haven’t, I ended up replacing board with nucleo and it
worked for my needs. I yesterday had an idea of checking out registers in a
running bluepill in debugger mode and copying what the register values are,
but I haven’t validated if that’s a good approach.