renode/renode #659: Authentication is needed to run `/usr/bin/mono' as the super user

GaurikaSohnvi, 2024-08-28:
### Description

Seeing the message to authenticate when trying to create a tap. When I try to authenticate, it always fails.
This issue goes away if I run renode as sudo.

### How to reproduce?

To reproduce, create a tap interface, as shown below:
```
➜  renode-source git:(01b2aff6) ✗ ./renode -d
15:09:48.2548 [WARNING] Couldn't start UI - falling back to console mode
15:09:48.5819 [INFO] Loaded monitor commands from: /home/gsohnvi/dev/renode-source/scripts/monitor.py
Renode, version 1.15.1.26765 (01b2aff6-202408281452)

(monitor) emulation CreateTap "tap" "tap1"
15:09:51.5298 [INFO] Copying...
==== AUTHENTICATING FOR org.freedesktop.policykit.exec ===
Authentication is needed to run `/usr/bin/mono' as the super user
Multiple identities can be used for authentication:

...

Choose identity to authenticate as (1-9): Invalid response `'.
```

### Environment

Please, provide the following information:

- OS: Ubuntu 20.04
- Renode version (with commit SHA): c0943c50000b9b81dcab96f029f975bf61278e2a

### Do you plan to address this issue and file a PR?

No

MateuszKarlic, 2024-08-29:
Hi @GaurikaSohnvi

Indeed there is a problem when providing input to `pkexec` while GUI is disabled. We will look into this issue, but for now the mitigation would be either:
* run Renode with `sudo` as you spotted
* run Renode with GUI enabled (you have to make sure that there is no `[WARNING] Couldn't start UI - falling back to console mode` in logs)
* configure sudo as passwordless
* create TAP interface beforehand, and connect Renode to an already existing interface. You will only need superuser privileges to create the interface, Renode should be able to connect without prompting for password. On Ubuntu you can do it like so:
```
[Remember to run all these commands as superuser]
ip tuntap add mode tap tap10
ip addr add X.X.X.X/YY dev tap10
ip link set tap10 up
```

Then in Renode just type:
```
(monitor) emulation CreateTap "tap10" "tap"
08:59:58.4539 [INFO] Opened interface tap10.
```

GaurikaSohnvi, 2024-08-29:
Thank you for the quick response @MateuszKarlic. GUI enabled seems to do the trick, thank you!

PiotrZierhoffer, 2024-09-19:
Hi @GaurikaSohnvi , I'll close this one for now. but if you have any additional comments - feel free to reopen.

For now we do not know if we'd be able to change the current behavior without UI, so the workaround with manual setting of TAP will probably become part of our docs.

closed 2024-09-19 by PiotrZierhoffer