renode/renode #664: Fix flash addresses in stm32h743.repl

GaurikaSohnvi, 2024-09-10:
### Description

Request to fix flash addresses in the stm32h743.repl, here's a patch:
```
diff --git a/platforms/cpus/stm32h743.repl b/platforms/cpus/stm32h743.repl
index ae1ab57b..369026dc 100644
--- a/platforms/cpus/stm32h743.repl
+++ b/platforms/cpus/stm32h743.repl
@@ -65,8 +65,8 @@ flashController: MTD.STM32H7_FlashController @ sysbus 0x52002000
     flash1: flash0
     flash2: flash1
 
-flash0: Memory.MappedMemory @ sysbus 0x80000000
-    size: 0x10000000
+flash0: Memory.MappedMemory @ sysbus 0x08000000
+    size: 0x100000
 
 // May be reserved
 sramBank2: Memory.MappedMemory @ sysbus 0x70000000
@@ -316,8 +316,8 @@ dtcm: Memory.MappedMemory @ sysbus 0x20000000
 ram: Memory.MappedMemory @ sysbus 0x1FF00000
     size: 0x20000
 
-flash1: Memory.MappedMemory @ sysbus 0x08000000
-    size: 0x200000
+flash1: Memory.MappedMemory @ sysbus 0x08100000
+    size: 0x100000
```

Reference: Table 7: memory map in the [reference manual](https://www.st.com/resource/en/reference_manual/dm00314099-stm32h742-stm32h743-753-and-stm32h750-value-line-advanced-arm-based-32-bit-mcus-stmicroelectronics.pdf)

### Do you plan to address this issue and file a PR?

No

mateusz-holenko, 2024-09-16:
HI @GaurikaSohnvi, thanks for reporting the problem.

The platform description is now fixed in https://github.com/renode/renode/commit/718dcdaa78b663b6644e57bd1da820f4a9907505. Hope this resolves your issue.

GaurikaSohnvi, 2024-09-16:
Thank you @mateusz-holenko!

closed 2024-09-16 by GaurikaSohnvi