renode/renode #665: Help required to write two external plugins

spulverail, 2024-09-12:
I am starting in using Renode in my project development but I am facing an issue. Maybe what I want to do is not allowed, so please help me figuring out what I am doing wrong.

### Description

My expectation is to implement Renode plugins using bare CSharp files in my own project. My problem is when I have two plugins `A.cs` and `B.cs`, and `A` is used as a constructor parameter of `B`. When doing this, I have a CS0234 error regarding `A` object in my `B` object.

### Usage example

I am just doing something like this in a simple terminal:

`$ ./renode-1.15.2/renode` 

and this in the Renode emulator:
```(monitor) logLevel -1
(monitor) verboseMode True
(monitor) include @./path/to/A.cs
Executing: i @/home/user/path/to/A.cs
(monitor) include @./path/to/B.cs
Executing: i @/home/user/path/to/B.cs
Errors during compilation or loading:
There were compilation errors:
/home/user/path/to/B.cs(20,60): error CS0234: The type or namespace name `A' does not exist in the namespace `Antmicro.Renode.Peripherals.Miscellaneous'. Are you missing an assembly reference?
```

For information, `A` is in `Antmicro.Renode.Peripherals.Timers` namespace, and `B` in `Antmicro.Renode.Peripherals.Miscellaneous`. I precise again that I am neither building these files (I let Renode doing it) or copying them inside the Renode sources folder.

Thanks for the help.

Mes-ser, 2024-09-12:
Just after  including `A.cs` run: 
> EnsureTypeIsLoaded "Antmicro.Renode.Peripherals.Timers.A"

Assuming your class name in `A.cs` is "A"

mateusz-holenko, 2024-09-13:
Hi @spulverail, thanks for asking the question!

Judging from the error message, .NET tries to locate the type `A` in the `Miscellaneous` namespace; you mentioned that it is in fact located in `Timers`, so perhaps you're missing a `using` or you should reference to `A` as `Timers.A`?

spulverail, 2024-09-13:
Hi @Mes-ser and @mateusz-holenko,
Thanks for the answer. I get the following error running the requested command:

```
(monitor) logLevel -1
(monitor) verboseMode True
(monitor) i @STM32L4_RTC.cs 
Executing: i @/home/user/path/to/STM32L4_RTC.cs
(monitor) EnsureTypeIsLoaded "AntMicro.Renode.Peripherals.Timers.STM32L4_RTC"
Executing: EnsureTypeIsLoaded "AntMicro.Renode.Peripherals.Timers.STM32L4_RTC"
There was an error executing command 'EnsureTypeIsLoaded "AntMicro.Renode.Peripherals.Timers.STM32L4_RTC"'
Unable to load the type `AntMicro.Renode.Peripherals.Timers.STM32L4_RTC`: Given type AntMicro.Renode.Peripherals.Timers.STM32L4_RTC was not found in any of the known assemblies.
```

I assume that if the command does not work, I cannot go further :(

mateusz-holenko, 2024-09-13:
Isn't it a typo in the namespace? Try `Antmicro.Renode.Peripherals.Timers.STM32L4_RTC` instead of `AntMicro.Renode.Peripherals.Timers.STM32L4_RTC`.

spulverail, 2024-09-17:
Indeed... It was first a typo error (thanks @mateusz-holenko) and it is now working when I use `Timers.A` instead of `A`.
Thanks for the help.