renode/renode #671: How do I set external IPs that can be accessed about RobotFramework HttpServer in windows?

tt2-woo, 2024-09-27:
### Description

`11:32:35.1301 [ERROR] Could not start the HTTP server on port 23800`

I hope to start HTTP server that some specific external IPs can be accessed. 

So I tryed to set listener.Prefixes.Add($"http://+:{port}/");

listener.Start() is not working. 

I also tried to set + or * below

```
netsh http add urlacl url=http://+:23800/ user=everyone
netsh http delete urlacl url=http://+:23800/

netsh http add urlacl url=http://*:23800/ user=everyone
netsh http delete urlacl url=http://*:23800/
```

but only "localhost" string can be applyed in windows. 

Isn't there any remote connecting(external IPs) way?

<renode-1.15.0/src/Renode/RobotFrameworkEngine/HttpServer.cs>
I wonder if external IPs are blocked in windows?
```
#if PLATFORM_WINDOWS
	listener.Prefixes.Add($"http://localhost:{port}/");
#else
	listener.Prefixes.Add($"http://*:{port}/");
#endif
```

### Expected behaviour

it can be applied `*` or `+` like linux(`listener.Prefixes.Add($"http://*:{port}/");`).

PiotrZierhoffer, 2024-10-03:
Hi @tt2-woo , thanks for reporting the issue.

Frankly speaking I'm not 100% sure how to address it. According to the [documentation](https://learn.microsoft.com/en-us/dotnet/fundamentals/runtime-libraries/system-net-httplistener) and some [SO discussions](https://stackoverflow.com/questions/4019466/httplistener-access-denied) this should, in general, work.

One remark I gather from that discussion is that the prefix must match exactly what is provided to `netsh`, and that there are OS-specific differences. On that, unfortunately, I won't be able to comment.

tt2-woo, 2024-10-07:
@PiotrZierhoffer Thanks for your reply.

In general, work including the usage about `netsh` and `+` in Windows. And Linux has no problem. Only Httpd in Renode about windows I have the trouble.

Could you handle with this problem next release?