renode/renode #677: STM32L562 HAL_RCC_ClockConfig going into Error_Handler()

Frisc98, 2024-10-14:
### Description
I've built a toy project (CubeMX generated) which uses UART4 to print out "Hello World" to uart. I've been trying to emulate it using renode, but I've run into issues regarding clock config. I've seen others have had the same issue but on their MCU issues seemed to be resolved.

Offending code is 
``` C
  if (HAL_RCC_ClockConfig(&RCC_ClkInitStruct, FLASH_LATENCY_5) != HAL_OK) {
    Error_Handler();
  }
```
.repl file I'm using is the one used in zephyr project for the same CPU.

```
partition_slot0partition_slot1nspartition_storagepartition: Memory.MappedMemory @ sysbus 0x0
    size: 0x4000000

flash0: Memory.MappedMemory @ sysbus 0x8000000
    size: 0x80000

sram0: Memory.MappedMemory @ sysbus 0x20000000
    size: 0x30000

timers2: Timers.STM32_Timer @ sysbus <0x40000000, +0x400>
    frequency: 110000000
    initialLimit: 0xffffffff
    ->nvic0@45

gpioa: GPIOPort.STM32_GPIOPort @ sysbus <0x42020000, +0x400>

gpiob: GPIOPort.STM32_GPIOPort @ sysbus <0x42020400, +0x400>

gpioc: GPIOPort.STM32_GPIOPort @ sysbus <0x42020800, +0x400>

gpiod: GPIOPort.STM32_GPIOPort @ sysbus <0x42020c00, +0x400>

gpioe: GPIOPort.STM32_GPIOPort @ sysbus <0x42021000, +0x400>

gpiof: GPIOPort.STM32_GPIOPort @ sysbus <0x42021400, +0x400>

gpiog: GPIOPort.STM32_GPIOPort @ sysbus <0x42021800, +0x400>

gpioh: GPIOPort.STM32_GPIOPort @ sysbus <0x42021c00, +0x400>

// autogenerated

redled9: Miscellaneous.LED @ gpiod 0x3
    invert: true

gpiod:
    3 -> redled9@0

greenled10: Miscellaneous.LED @ gpiog 0xc
    invert: true

gpiog:
    12 -> greenled10@0

nvic0: IRQControllers.NVIC @ {
    sysbus new Bus.BusPointRegistration { address: 0xe000e000; cpu: cpu0 }
}
    -> cpu0@0
    systickFrequency: 110000000

cpu0: CPU.CortexM @ sysbus
    cpuType: "cortex-m33"
    nvic: nvic0

adc1: Analog.STM32_ADC @ sysbus 0x42028000
    IRQ->nvic0@37

i2c1: I2C.STM32F7_I2C @ sysbus 0x40005400
    EventInterrupt->nvic0@55
    ErrorInterrupt->nvic0@56

rcc: Python.PythonPeripheral @ sysbus 0x40021000
    size: 0x400
    initable: true
    filename: "scripts/pydev/flipflop.py"

rng: Miscellaneous.STM32F4_RNG @ sysbus 0x420c0800
    ->nvic0@94

rtc: Timers.STM32F4_RTC @ sysbus 0x40002800
    AlarmIRQ->nvic0@2

spi1: SPI.STM32SPI @ sysbus 0x40013000
    IRQ->nvic0@59

spi3: SPI.STM32SPI @ sysbus 0x40003c00
    IRQ->nvic0@99

usart1: UART.STM32F7_USART @ sysbus 0x40013800
    frequency: 200000000
    IRQ->nvic0@61

usart3: UART.STM32F7_USART @ sysbus 0x40004800
    frequency: 200000000
    IRQ->nvic0@63

usart4: UART.STM32F7_USART @ sysbus 0x40004C00
    frequency: 110000000
    IRQ -> nvic0@64
// cortex-m overlay

dwt: Miscellaneous.DWT @ sysbus 0xE0001000
    frequency: 72000000

// st,stm32l5 overlay

sysbus:
    init:
        Tag <0x40007000, 0x400073FF> "pwr" 0xFFFFFFFF

```

Renode logs show entering that function after reading pwr tag 
```
15:38:12.0905 [INFO] machine-2/cpu0: Entering function SystemClock_Config at 0x80003FA
15:38:12.0906 [INFO] machine-2/cpu0: Entering function SystemClock_Config at 0x8000400
15:38:12.0906 [INFO] machine-2/cpu0: Entering function HAL_PWREx_ControlVoltageScaling (entry) at 0x8000AC4
15:38:12.0906 [WARNING] machine-2/sysbus: [cpu0: 0x8000AC4] (tag: 'pwr') ReadDoubleWord from non existing peripheral at 0x40007000, returning 0xFFFFFFFF.
15:38:12.0906 [INFO] machine-2/cpu0: Entering function HAL_PWREx_SMPS_GetEffectiveMode (entry) at 0x8000AA0
15:38:12.0906 [WARNING] machine-2/sysbus: [cpu0: 0x8000AA0] (tag: 'pwr') ReadDoubleWord from non existing peripheral at 0x40007010, returning 0xFFFFFFFF.
15:38:12.0907 [INFO] machine-2/cpu0: Entering function HAL_PWREx_SMPS_GetEffectiveMode at 0x8000AB8
15:38:12.0907 [INFO] machine-2/cpu0: Entering function HAL_PWREx_SMPS_GetEffectiveMode at 0x8000AB0
15:38:12.0907 [INFO] machine-2/cpu0: Entering function HAL_PWREx_ControlVoltageScaling at 0x8000AD0
15:38:12.0907 [INFO] machine-2/cpu0: Entering function HAL_PWREx_ControlVoltageScaling at 0x8000AD6
15:38:12.0907 [WARNING] machine-2/sysbus: [cpu0: 0x8000AD6] (tag: 'pwr') ReadDoubleWord from non existing peripheral at 0x4000700C, returning 0xFFFFFFFF.
15:38:12.0908 [INFO] machine-2/cpu0: Entering function HAL_PWREx_ControlVoltageScaling at 0x8000B36
15:38:12.0908 [INFO] machine-2/cpu0: Entering function HAL_PWREx_ControlVoltageScaling at 0x8000B30
15:38:12.0908 [INFO] machine-2/cpu0: Entering function SystemClock_Config at 0x800040A
15:38:12.0908 [INFO] machine-2/cpu0: Entering function SystemClock_Config at 0x800045E
15:38:12.0908 [INFO] machine-2/cpu0: Entering function Error_Handler (entry) at 0x80003E8
15:38:12.0909 [INFO] machine-2/cpu0: Entering function Error_Handler at 0x80003EA
```

Has anyone got any clues as to how to debug this or has any workarounds? If you need any more info, just let me know.

Frisc98, 2024-10-16:
To pass all the power checks, I had to define following registers:
```
sysbus:
    init:
        Tag <0x40007000, 0x400073FF> "pwr" 0xFFFFFFFF
        Tag <0x40022000 4> "FLASH_ACR" 0x5
        Tag <0x40021008 4> "RCC_CFGR" 0x4
        Tag <0x4000700C 4> "PWR_CR4" 0xFFFF3FFF
        Tag <0x40007014 4> "PWR_SR2" 0x00000000
```
This leads to one final failing function ```HAL_InitTick(uwTickPrio)``` in ```HAL_RCC_ClockConfig()```.
It fails on this line: 
```
      /* Configure the SysTick IRQ priority */
      if (TickPriority < (1UL << __NVIC_PRIO_BITS))
      {
        HAL_NVIC_SetPriority(SysTick_IRQn, TickPriority, 0U);
        uwTickPrio = TickPriority;
      }
      else
      {
        status = HAL_ERROR;
      }
``` 
because TickPriority is the same as 1 << NVIC_PRIO_BITS (both are equal to 8 in my case).

tyler-macinnis, 2024-10-16:
If you look in [stm32g0.repl](https://github.com/renode/renode/blob/master/platforms/cpus/stm32g0.repl) it defines RCC with python, you might be able to achieve something similar. Either that defining your own custom `rcc` peripheral.

Frisc98, 2024-10-17:
I've looked at your link, python script was extremely helpful. 
Had to change ``` request.value = data['sw'] << 3``` to correct offset for stm32l5xx to ``` request.value = data['sw'] << 2```.
Had to add following to python script for rcc: 
```
    elif request.offset == 0x88:
        request.value = 0x00000080
```
Had to add these registers:
```
sysbus:
    init:
        Tag <0x40007000, 0x400073FF> "pwr" 0xFFFFFFFF
        Tag <0x40021000 4> "RCC_CR" 0x3F020063
        Tag <0x40021088 4> "RCC_CCIPR1" 0x00008000
        Tag <0x4000700C 4> "PWR_CR4" 0xFFFF3FFF
        Tag <0x40007014 4> "PWR_SR2" 0x00000000
```
Best way to deal with all of this is to setup debugging (used this as a guide https://medium.com/@pc0is0me/getting-started-with-stm32f4-emulation-using-renode-f6cb158d27d1). Step through the code until you hit Error_Handler(), change register values in rcc python script or with tags, rinse, repeat until it works.

Full .repl file for reference:
```
partition_slot0partition_slot1nspartition_storagepartition: Memory.MappedMemory @ sysbus 0x0
    size: 0x4000000

flash0: Memory.MappedMemory @ sysbus 0x8000000
    size: 0x80000

flash_ctrl: MTD.STM32F4_FlashController @ sysbus 0x40022000
    flash: flash0

sram0: Memory.MappedMemory @ sysbus 0x20000000
    size: 0x30000

timers2: Timers.STM32_Timer @ sysbus <0x40000000, +0x400>
    frequency: 110000000
    initialLimit: 0xffffffff
    ->nvic0@45

gpioa: GPIOPort.STM32_GPIOPort @ sysbus <0x42020000, +0x400>

gpiob: GPIOPort.STM32_GPIOPort @ sysbus <0x42020400, +0x400>

gpioc: GPIOPort.STM32_GPIOPort @ sysbus <0x42020800, +0x400>

gpiod: GPIOPort.STM32_GPIOPort @ sysbus <0x42020c00, +0x400>

gpioe: GPIOPort.STM32_GPIOPort @ sysbus <0x42021000, +0x400>

gpiof: GPIOPort.STM32_GPIOPort @ sysbus <0x42021400, +0x400>

gpiog: GPIOPort.STM32_GPIOPort @ sysbus <0x42021800, +0x400>

gpioh: GPIOPort.STM32_GPIOPort @ sysbus <0x42021c00, +0x400>

// autogenerated

redled9: Miscellaneous.LED @ gpiod 0x3
    invert: true

gpiod:
    3 -> redled9@0

greenled10: Miscellaneous.LED @ gpiog 0xc
    invert: true

gpiog:
    12 -> greenled10@0

nvic0: IRQControllers.NVIC @ {
    sysbus new Bus.BusPointRegistration { address: 0xe000e000; cpu: cpu0 }
}
    -> cpu0@0
    systickFrequency: 110000000

cpu0: CPU.CortexM @ sysbus
    cpuType: "cortex-m33"
    nvic: nvic0

adc1: Analog.STM32_ADC @ sysbus 0x42028000
    IRQ->nvic0@37

i2c1: I2C.STM32F7_I2C @ sysbus 0x40005400
    EventInterrupt->nvic0@55
    ErrorInterrupt->nvic0@56

rcc: Python.PythonPeripheral @ sysbus 0x40021000
    size: 0x400
    initable: true
    script: '''
if request.isInit:
   lastVal = 0
   data = {'hsion': 1, 'pllon': 0, 'sw': 0, 'lsion': 0}
self.NoisyLog("%s on FLIPFLOP at offset 0x%x, value 0x%x, length %d, %s" % (str(request.type), request.offset, request.value, request.length, str(data)))

if request.isWrite:
    if request.offset == 0x0:
        # HSION [doc = "Bit 8 - HSI16 clock enable"]
        data['hsion'] = (request.value >> 8) & 0x1
        # PLLON [doc = "Bit 24 - PLL enable"]
        data['pllon'] = (request.value >> 24) & 0x1
    elif request.offset == 0x8:
        # SW_W [doc = "Bits 0:2 - System clock switch"]
        data['sw'] = request.value & 0x7
    elif request.offset == 0x60:
        data['lsion'] = request.value & 0x1
elif request.isRead:
    lastVal = 1 - lastVal
    if request.offset == 0x0:
        # request.value = lastVal * 0x83

        # RCC_CR [doc = "0x00 - Clock control register"]
        # HSION_RW [doc = "Bit 8 - HSI16 clock enable flag"]
        # HSIRDY_R [doc = "Bit 10 - HSI16 clock ready flag"] # HSIRDY == HISON for this model
        # PLLRDY [doc = "Bit 25 - PLL clock ready flag"]
        request.value = (data['hsion'] << 8) | (data['hsion'] << 10) | (data['pllon'] << 25)
    elif request.offset == 0x4:
        request.value = lastVal * 0xFFFFFFF8
    elif request.offset == 0x24:
        request.value = 0x0
    elif request.offset == 0x60:
        request.value = (data['lsion'] << 1)
    elif request.offset == 0x88:
        request.value = 0x00000080
    else:
        # RCC_CFGR [doc = "0x08 - Clock configuration register"]
        # SWS_R [doc = "Bits 3:5 - System clock switch status"]
        # request.value = lastVal * 0xFFFFFFFF
        request.value = data['sw'] << 2

self.NoisyLog("  0x%x (%s)" % (request.value, str(data)))
'''

rng: Miscellaneous.STM32F4_RNG @ sysbus 0x420c0800
    ->nvic0@94

rtc: Timers.STM32F4_RTC @ sysbus 0x40002800
    AlarmIRQ->nvic0@2

spi1: SPI.STM32SPI @ sysbus 0x40013000
    IRQ->nvic0@59

spi3: SPI.STM32SPI @ sysbus 0x40003c00
    IRQ->nvic0@99

usart1: UART.STM32F7_USART @ sysbus 0x40013800
    frequency: 200000000
    IRQ->nvic0@61

usart3: UART.STM32F7_USART @ sysbus 0x40004800
    frequency: 200000000
    IRQ->nvic0@63

usart4: UART.STM32F7_USART @ sysbus 0x40004C00
    frequency: 110000000
    IRQ -> nvic0@64
// cortex-m overlay

dwt: Miscellaneous.DWT @ sysbus 0xE0001000
    frequency: 72000000

// st,stm32l5 overlay

sysbus:
    init:
        Tag <0x40007000, 0x400073FF> "pwr" 0xFFFFFFFF
        Tag <0x40021000 4> "RCC_CR" 0x3F020063
        Tag <0x40021088 4> "RCC_CCIPR1" 0x00008000
        Tag <0x4000700C 4> "PWR_CR4" 0xFFFF3FFF
        Tag <0x40007014 4> "PWR_SR2" 0x00000000
```

closed 2024-10-17 by Frisc98