renode/renode #679: Assistance setting a timestamp as part of a log file's name from a .resc script

tyler-macinnis, 2024-10-16:
### Description

I would appreciate guidance on setting a timestamp for a log name in a Renode Script (likely via Python).

### Usage example

A usage example would be running a `.resc` script and it sets a timestamp as part of the log file name.

### Additional information

For robot tests, I can easily set a timestamp like this.

```robot
# Filename: renode-test-flash-size.robot
*** Settings ***
Suite Setup       Setup
Suite Teardown    Teardown
Test Setup        Reset Emulation
Resource          ${RENODEKEYWORDS}
Library           DateTime

*** Variables ***
${PLATFORM}              ${CURDIR}/../Platforms/MyPlatform.repl
${CUSTOM_PERIPHERALS}    ${CURDIR}/../Scripts/renode-custom-peripherals.resc
${ELF}                   ${CURDIR}/../../Debug/MyElf.elf
${SVD}                   ${CURDIR}/../SVD/STM32G071.svd
${FLASH}                 sysbus.flash0

*** Keywords ***
Create Machine
    Execute Command    i @${CUSTOM_PERIPHERALS}
    Execute Command    mach create
    Execute Command    machine LoadPlatformDescription @${PLATFORM}
    ${date} =          Get Current Date                                result_format=%Y-%m-%d_%H-%M-%S
    ${LOGOUTPUT}       Set Variable                                    ${CURDIR}/../Logs/renode-test-template-${date}.log
    Execute Command    logLevel 1
    Execute Command    logFile @${LOGOUTPUT}
    Execute Command    logLevel -1 file
    Execute Command    sysbus.cpu0 LogFunctionNames true
    Execute Command    sysbus LogAllPeripheralsAccess true
    Execute Command    sysbus LoadELF @${ELF}
    Execute Command    sysbus ApplySVD @${SVD}

*** Test Cases ***
Basic Test to Access Flash
    Create Machine
    Start Emulation
    Sleep              3
    ${FlashSize}=      Execute Command         ${FLASH} Size
    Should Be Equal    ${FlashSize.strip()}    0x0000000000020000
```

How can I do something similar with **ONLY** a .resc script?

Here is my current `.resc` script as an example

```
:name: renode-config-debug
:description: This script serves as an example and starts renode for a debug session

# The ?= notation indicates a default value if the variable is not set.
$bin ?= $ORIGIN/../../Debug/MyElf.elf
$svd ?= $ORIGIN/../SVD/STM32G071.svd
$platform ?= $ORIGIN/../Platforms/MyPlatform.repl
$customPeripherals ?= $ORIGIN/renode-custom-peripherals.resc
$logLocation ?= $ORIGIN/../Logs/renode-config-debug.log

i $customPeripherals

mach create
machine LoadPlatformDescription $platform

# Logging Setup
# Set the verbosity NOISY (-1), DEBUG (0), INFO (1), WARNING (2), ERROR (3).
logLevel 1
logFile $logLocation
logLevel -1 file

sysbus.cpu0 LogFunctionNames true
sysbus LogAllPeripheralsAccess true

macro reset
"""
    sysbus LoadELF $bin
    sysbus ApplySVD $svd
"""

runMacro $reset

machine StartGdbServer 3333
```

### Do you plan to address this issue and file a PR?

I've been struggling to get this to work for a while now. I would appreciate some support because I imagine it's fairly simple to do, I just don't know how.

tyler-macinnis, 2024-10-25:
I found a way to do this myself.

Essentially I have a `renode-custom-peripherals.resc` that includes my custom peripherals and commands.

Inside of it I have

```
python
"""
from datetime import datetime
import os

log_dir = '$global.ORIGIN/../Logs/'

def mc_timestampedlog(name):
    timestamp = datetime.now().strftime('%Y-%m-%d_%H-%M-%S')
    self.Parse('logFile '+ log_dir + name + '-' + timestamp + '.log')
"""
```

This way in my `.resc` files I can simply do this:

```
$platform ?= $ORIGIN/../Platforms/myboard.repl
$customPeripherals ?= $ORIGIN/renode-custom-peripherals.resc
$logName ?= renode-config-debug

i $customPeripherals

mach create
machine LoadPlatformDescription $platform

# Logging Setup
# Set the verbosity NOISY (-1), DEBUG (0), INFO (1), WARNING (2), ERROR (3).
logLevel 1
timestampedlog $logName
logLevel -1 file
```

Can also include the `renode-custom-peripherals.resc` in a `.robot` test and use it in a similar way.

closed 2024-10-25 by tyler-macinnis

nrabulinski, 2024-10-28:
Thanks for opening an issue! Currently there's no better and/or built-in way for doing this, and your solution is pretty much exactly what we would've suggested