renode/renode #680: Call function from imported python script in UART hook

tomhepworth, 2024-10-17:
### Description

I import a python script in my .resc file  with `import @path/to/file.py`
```
def my_function(data):
    print("Hello World")
    return
    
print("Python file loaded successfully")
```

As the resc file is run I see `Python file loaded successfully` in the monitor.  I can also use `python  "my_function()"` from the monitor.

When I run `uart AddLineHook "string" "my_function()` and trigger this hook, I get:
```
[ERROR] uart: Python runtime error: name 'my_function' is not defined
``` 

This also happens for `AddAfterReadDoubleWordHook`, but I have not tested other hook commands. `AddAfterReadDoubleWordHook` and others also appear to be totally undocumented?

My intention is to intercept data packets sent over uart and format the packets into readable strings. Is there a better way? 

 
### Usage example

This should work:
`uart AddLineHook "line" "my_python_function(...)`

 
### Additional information

NA

### Do you plan to address this issue and file a PR?

Possibly

tomhepworth, 2024-10-17:
I was able to achieve the behaviour I want with 

```
def my_match(line):
    return "string" in line

def my_function(line):
    print("Hello World")
    return True

Antmicro.Renode.Hooks.UartHooksExtensions.AddLineHook(monitor.Machine["sysbus.uart"], my_match, my_function)
````

It does seem like the most intuitive method would be the one I describe in my issue though.

nrabulinski, 2024-10-30:
Hey @tomhepworth, thanks for reporting this.
You're right that using a Python function defined in the monitor with `AddLineHook`, `AddReadHook`, `InstallCustomInstructionHandlerFromString`, etc, is impossible. The reason is that each of those methods creates its own `PythonEngine` instance, so effectively all of those scripts are isolated from each other and you should think about them as such.

This means that for now if you want to use a Python function defined in the Monitor, or share state between Python hooks, you need to install the hook either from a Python script, or by using the Monitor's `python` command, exactly like you managed to work it out yourself.

For a slightly more self-contained notation, but probably not that useful in practice, you can do something like this as well:

```
(machine-0) set var """
def my_function(data):
    print(data)

my_function(line)
"""
(machine-0) uart AddLineHook "string" $var
```
It will redefine the function every time you trigger the hook though, so that's why it's not as flexible as you'd possibly want.

tomhepworth, 2024-10-30:
Thanks for the reply

It makes sense but I think the behaviour and workaround should be better documented, as it  is slightly counter intuitive right now.