renode/renode #682: Identifying delay() in the running target binary and skip it

asmitaj08, 2024-10-23:
Hello,
I am using Renode using Python script via `pyrenode3`. I want to know if there is  a way to identify if any delay() or while(1); for(;;;) being executed in running target and skip it?
I did try to add CPU Hook at those addresses, but it's not working correctly.
For example, in one of the target binary, the delay() is at address `0x4ec0` , I added cpu hook at this address, withing cpuHook, I paused the machine, modified the PC , and resumed, but it won't work in correct state. I understand it might be happening because of the other status flag being set and register state being changed which might not be recovered just by modifying PC. Hence, I wanted to know if there is any other, anything that I can modify internally in Renode to make it work or any other suggestion?

I am attaching some screenshots for reference :
[image]

Couple of things that I tried within Hook were : 
First :   
```
mach.Pause()
mach.sysbus.cpu.SetRegisterUnsafe(15, RegisterValue.Create((delay_addr+4), 32)) #delay_addr is 0x4ec0
mach.Start()
```

Second :  
```
mach.Pause()
mach.sysbus.cpu.SetRegisterUnsafe(15, RegisterValue.Create((delay_addr+4), 32)) #delay_addr is 0x4ec0
mach.sysbus.cpu.SetRegisterUnsafe(0, RegisterValue.Create(0, 32))
mach.Start()
```
Third :  
```
mach.Pause()
mach.sysbus.cpu.SetRegisterUnsafe(15, RegisterValue.Create((mach.sysbus.cpu.GetRegisterUnsafe(14)), 32)) #for lr
mach.sysbus.cpu.SetRegisterUnsafe(0, RegisterValue.Create(0, 32))
mach.Start()
```

Thank you!

mateusz-holenko, 2024-11-21:
Hi @asmitaj08,

To avoid the necessity of providing the symbol offset directly (and thus to make your script more reusable between compilations of the simulated software) you can obtain it at runtim using:
```
offset = mach.sysbus.GetSymbolAddress("name_of_interesting_symbol")
```

When it comes to skipping the delay functions, Renode provides a basic mechanism for this - check the `SkipTime` method provided by CPU and the `ZephyrMode` plugin as an example of its usage (please not the even though this implementation targets the Zephyr RTOS, the idea of skipping time is generic enough to be used with different software).

Also - you don't have to pause machine in order to change CPU PC from a hook. The hook itself is executed from the CPU thread and it's blocking (CPU does not actively execute any instructions while the hook is being handled).

asmitaj08, 2024-11-21:
Thank you so much for your response. It's really helpful. If you don't mind, can I ask a different question that I have been struggling with , otherwise I can open a new thread for it.

Question : So, far to take the screenshot and reload it , I have been using `Save` and `Load` feature of Renode. However, `Load` takes about 2 seconds (As I am calling it inside every fuzz run, it slows it down a lot), I was wondering if there is any other way that I can keep track of the every memory, regiter and status flag that is being changed and then restore the original state  during start of every fuzz run.

So, the goal is, I first execute the emulator until it reaches the target function, then I save the state using `Save` , then at start of every fuzz loop, I load this original state using `Load`. And it works completely fine, just that it makes fuzzer quite slow. Hence I was looking for alternate approach. Your suggestions would be valauable, thank you!

asmitaj08, 2025-03-21:
Thank you! This is very useful feature.

closed 2025-03-21 by asmitaj08