renode/renode #695: Control and Status Registers support -RISC-V

Maleehaakbar, 2024-11-08:
### Description

I work on custom Andes chip and emulate interrupts using CLIC in Renode in non-vectored CLIC mode.
However , Renode have incomplete support of MTVT ,MNXTI  and other control and status registers for RISCV.
Will renode plan provide support for hardware vectoring mode of CLIC and related CSRs in near future?

If not, kindly give some guidelines on how I provide support for these registers.

p-woj, 2024-11-21:
Hi, could you provide some details about which CSRs you mean? We believe all of the standard ones from [the proposed specification](https://github.com/riscv/riscv-fast-interrupt) should be implemented - do you mean some nonstandard registers like `MTVT2`?

It would also be very helpful if you could share some software that fails due to the incompletely supported registers.

Maleehaakbar, 2024-11-21:
The support of mtvt is present in tlib submodule but it is not present outside the tlib submodule.
For exsample:
`#define CSR_MTVT                          0x307 /* unratified as of 2024-06; smclic extension */`

So, at the backend mtvt CSR support is present, however in C# at the front end, there is no support and because of this, the code fails in interrupt code of zephyr and CPU aborts.
[image]

Maleehaakbar, 2024-12-03:
Hi, I open the PR on both repositories. 
Following are the links . 
https://github.com/antmicro/tlib/pull/17
https://github.com/renode/renode-infrastructure/pull/119