renode/renode #696: How to trigger FinishTransmission in SPI?

naitgacem, 2024-11-09:
### Description

An SPI peripheral is not behaving correctly (ENC28J60) due to (I am guesstig here) burst mode. Once i read from buffer memory once, all subsequent commands are treated as read buffer memory.
This method is never called. Confirmed by adding logging statement.
```C#
public void FinishTransmission()
        {
            this.Log(LogLevel.Info, "Transmission finished!!!!!!!!!!!!!!!.");
            lock(sync)
            {
                currentMode = Mode.Normal;
            }

```}

mateusz-holenko, 2024-11-21:
Hi @naitgacem,

In Renode it is the SPI controller responsibility to call `FinishTransmission` on the SPI peripheral (see https://github.com/renode/renode-infrastructure/blob/master/src/Emulator/Peripherals/Peripherals/SPI/MPFS_SPI.cs#L191 for reference).

You should check implementation of the controller you use in the platform.

naitgacem, 2024-11-21:
I had this issue with the STM32 SPI controller. It doesn't seem to call that function. Thanks for the pointer in the correct direction. I think I will close this for now.

closed 2024-11-21 by naitgacem