renode/renode #701: Make `cluster` interface more aligned with the `cpu` interface

emmedve1, 2024-11-14:
### Description

For consistency's sake, can [SetIsHalted](https://github.com/renode/renode-infrastructure/blob/01ab6418585e9ee29482263d61b5cee184bf2a15/src/Emulator/Peripherals/Peripherals/CPU/Cluster.cs#L19-L25):
```c#
        public static void SetIsHalted(this Cluster cluster, bool value)
        {
            foreach(var cpu in cluster.Clustered)
            {
                cpu.IsHalted = value;
            }
        }
```
be renamed to [IsHalted](https://github.com/renode/renode-infrastructure/blob/01ab6418585e9ee29482263d61b5cee184bf2a15/src/Emulator/Peripherals/Peripherals/CPU/CPUCore.cs#L67) to match the property/"command" for individual `cpu`(s):

### Usage example

Currently for a `cpu` can be stopped/started individually like this:

```
    cpu IsHalted False
```

For a `cluster` it is:
```
    cluster SetIsHalted False
```

Would be nice to be:

```
    cluster IsHalted False
```

emmedve1, 2024-11-14:
@kknox -> participant

FranciszekPin, 2024-12-17:
Hi @emmedve1, thanks for the suggestion. We changed the halting API for clusters, now it's consistent with the API for CPU. Now `IsHalted` should be called:

```
    cluster IsHalted False
```

Please keep in mind that only a setter is available, because a getter value would be hard to specify (when some CPUs are halted and some not).