renode/renode #705: Can we have a binary build for aarch64?

Eplankton, 2024-11-24:
Would like to build Renode emulator from the source code on a raspberrypi which is based on a arm64 broadcom chip, so I guess it may not be so diffcult to be done. But things got weird when some errors began to compliant about `impossible constraints of __asm__`  and the `-Werror=clobberred` etc.

Eplankton, 2024-12-11:
> Hi @Eplankton
> 
> You can build it with `./build.sh --net --host-arch aarch64` - but please keep in mind that
> 
> a) it requires dotnet to be installed on host b) this is experimental and not complete. You can expect progress in our coverage of missing features, but at this moment you will have to assume that things don't fully work yet.

It seems to be an unknown error.

LillyJinstrand, 2025-01-09:
Hi @Eplankton and @Citrus-Marmalade, there is now much more complete aarch64 support available on this branch https://github.com/renode/renode/tree/68189-aarch64-tcg-ops. It is still experimental, but you should not encounter any tcg errors with this version.

Citrus-Marmalade, 2025-01-15:
> Hi @Eplankton and @Citrus-Marmalade, there is now much more complete aarch64 support available on this branch https://github.com/renode/renode/tree/68189-aarch64-tcg-ops. It is still experimental, but you should not encounter any tcg errors with this version.

Hello @LillyJinstrand, 
I can confirm that switching to this branch fixed our issue.

Eplankton, 2025-01-16:
> Hello @LillyJinstrand, I can confirm that switching to this branch fixed our issue.

Can we build the renode with cross-compile toolchain and move the binary to aarch64 targets? Due to the low performance of raspberrypi, it will be diffcult to build on it.

Eplankton, 2025-05-09:
Now we have the update, it seems that native build on `aarch64` has been released according to this official blog: https://renode.io/news/native-64-bit-arm-host-support-in-renode/

closed 2025-05-09 by Eplankton