renode/renode #707: Debug a bogus "Access out of bounds" Exception being thrown

naitgacem, 2024-11-25:
### Description

Running an example involving an SPI peripheral. Using one node works stable enough. Adding more nodes (involving networking) crashes sometimes with an index out of bounds exception. I have added log statements before and after every array access and there does not seem to be any access just before the exception was thrown.

I would like to figure if this is an issue with the simulation model or with Renode framework itself.
Where does one start to pinpoint this issue?

PiotrZierhoffer, 2024-11-25:
Hi @naitgacem,

Thanks for the report.

It's difficult to say based on a vague description. 

I believe the best way to start the discussion would be for you to create a reproduction case that we could look at, something like https://github.com/renode/renode-issue-reproduction-template (you can use it as a template).

Also just pasting the exception itself would help.

naitgacem, 2024-11-26:
Hi @PiotrZierhoffer, thanks for getting back to me with this.
So it is indeed vague because I was hoping I could track it down myself, since I can't reproduce this unless I spin up multiple MCUs isnstances, one works just fine.
However the exception is pasted here below:
Fatal error:

> Fatal error:
> Index was outside the bounds of the array.
>   at Antmicro.Renode.Peripherals.Network.KS8851MOD.Transmit(Byte data)
>   at Antmicro.Renode.Peripherals.SPI.STM32SPI.HandleDataWrite(UInt32 value)
>   at Antmicro.Renode.Core.Structure.Registers.PeripheralRegister.RegisterField`1.CallWriteHandler(UInt64 oldValue, UInt64 newValue)
>   at Antmicro.Renode.Core.Structure.Registers.PeripheralRegister.WriteInner(Int64 offset, UInt64 value)
>   at Antmicro.Renode.Core.Structure.Registers.BaseRegisterCollection`2.TryWrite(Int64 offset, T value)
>   at Antmicro.Renode.Core.Structure.Registers.BaseRegisterCollection`2.Write(Int64 offset, T value)
>   at Antmicro.Renode.Peripherals.SPI.STM32SPI.WriteByte(Int64 offset, Byte value)
>   at Antmicro.Renode.Peripherals.Bus.SystemBus.WriteByte(UInt64 address, Byte value, ICPU context)
>   at Antmicro.Renode.Peripherals.CPU.TranslationCPU.WriteByteToBus(UInt64 offset, UInt64 value)
>   at WriteByteToBusWrapper(CortexMWrappers , UInt64 , UInt64 )
>--- End of stack trace from previous location where exception was thrown ---
>   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
>   at Antmicro.Renode.Utilities.Binding.ExceptionKeeper.ThrowExceptions()
>   at TlibExecute(CortexMWrappers , Int32 )
>   at Antmicro.Renode.Peripherals.CPU.TranslationCPU.ExecuteInstructions(UInt64 numberOfInstructionsToExecute, UInt64& numberOfExecutedInstructions)
>   at Antmicro.Renode.Peripherals.CPU.BaseCPU.CpuThreadBodyInner(Boolean singleStep)
>   at Antmicro.Renode.Peripherals.CPU.BaseCPU.CpuThreadBody()
>   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
>   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
>   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)
>   at System.Threading.ThreadHelper.ThreadStart()

PiotrZierhoffer, 2024-11-27:
Am I right to assume you have your own version of KS8851?

Please keep in mind that our model of KS8851 is quite old and incomplete. Of course we don't want it to crash, but I see some logic in the `Transmit` function that could misbehave depending on the input data.

Still, this will be fairly difficult to decipher without a reproduction case.

naitgacem, 2024-11-27:
Yes I had my own version. However after adding extensive logging and building from source I was able to track it to my own code. So I'm closing this for now.
----
By the way, the build script is broken on windows using dotnet and cygwin. To build, I had first to hardcode $ROOT_PATH into a windows path rather than a Cygwin one (CMake complained). 

I also had to bump all `LangVersion` from 7.2 to 9.0 to accommodate the error `Module initializers` requires 9.0 or higher.