renode/renode #708: Latest nightly releases are unable to run robot tests

nicolak95, 2024-11-27:
## Overview

Just to keep you informed, the last few nightly releases from https://builds.renode.io/ have been failing when trying to run Robot tests.

Error -> "location was unexpected at this time"

[image]

## Environment

I am running tests on a windows machine using VS Code.

The versions tested where this issue was seen:
renode_1.15.3+20241125gitfb0ff392d
renode_1.15.3+20241122gitc2005c065
renode_1.15.3+20241120git93842864c

This issue could go back further but I have not attempted further investigation. 
I tested a release also from August which had no issues.

PiotrZierhoffer, 2024-11-28:
Hi @nicolak95 

Since you're running our tests via a wrapper, it's not exactly clear where this stems from.

The `location was unexpected at this time` error seems to be generated by CMD/PowerShell itself.

Can you extract the command from your wrapper that actually run renode-test?

nicolak95, 2024-11-28:
Hi Piotr, strangely I have not been able to replicate the issue I was seeing yesterday with the same fw releases. So for now I will resolve this thread. Thanks

closed 2024-11-28 by nicolak95