renode/renode #709: Error E39: Memory size has to be aligned to guest page size

godveryday, 2024-11-27:
I created and ran a repl file based on leon3, but the following error occurred:

error e39: exception was thrown during registratin of 'cpu' in 'sysbus' memory size has to be aligned to guest page size

My repl file is below:

```c
sysbus:
    Endianess: Endianess.BigEndian

rom: Memory.MappedMemory @ sysbus 0x00000000
     size: 0x80000

ram: Memory.MappedMemory @ sysbus 0x40000000
    size: 0x400000

cpu: CPU.Sparc @ sysbus
    cpuType: "leon3"

``` 

I don't know what's going on, please help me

PiotrZierhoffer, 2024-11-27:
Hi @godveryday 

Apart from the additional space near the `     size: 0x80000` line - this works fine for me.

Can you please double check on the latest Renode?

godveryday, 2024-11-28:
I want to know the cause of why this error occurs.
I'll show you my entire code, so can you tell me the solution and the cause?

```
sysbus:
	Endianess: Endianess.BigEndian

rom: Memory.MappedMemory @ sysbus 0x00000000
	size: 0x80000

edac: Memory.MappedMemroy @ sysbus 0x40000000
	size: 0x400000

io_area: Memory.MappedMemory @ sysbus 0x20000000
	size: 0x218

reg: Memory.MappedMemory @ sysbus 0x20000220
	size: 0x14

gp_reg: Memory.MappedMemory @ sysbus 0x20000240
	size: 0xC

... 

uart0: UART.GaislerAPBUART @ sysbus <0x80000100, +0x100>

cpu: CPU.Sparc @ sysbus
	cpuType: "leon3"

```

PiotrZierhoffer, 2024-11-28:
Hi @godveryday 

The offending part is 
```
io_area: Memory.MappedMemory @ sysbus 0x20000000
	size: 0x218

reg: Memory.MappedMemory @ sysbus 0x20000220
	size: 0x14

gp_reg: Memory.MappedMemory @ sysbus 0x20000240
	size: 0xC
```

In Renode your MappedMemory blocks have to be aligned to guest page boundaries.

In the case of SPARC it is 0x1000, so both the registration address and the size has to be dividable by 0x1000. Probably the easiest approach would be to have one big region for this address range.

We are improving the error message to be more descriptive.

godveryday, 2024-11-29:
Where can i get this information
"SPARC it is 0x1000"

PiotrZierhoffer, 2024-11-30:
With the latest Renode you will get this information printed out in the log (we have added the message prompted by your issue).

There is no 0x1000 const anywhere, it's calculated from some parameters in the https://github.com/antmicro/tlib submodule.