renode/renode #710: RESC issue regarding plugin inclusion

spulverail, 2024-11-27:
Hello there,

Few months ago, you helped my intern and me to resolve a namespace issue (#665).
Please find just below my `.resc` file:

```using sysbus
mach create "Test"

i @../MyRND/MyPeripherals/STM32L4_DBGMCU.cs
i @../MyRND/MyPeripherals/STM32L4_FlashController.cs
i @../MyRND/MyPeripherals/STM32L4_FlashSize.cs
i @../MyRND/MyPeripherals/STM32L4_PWR.cs
i @../MyRND/MyPeripherals/STM32L4_UARTbis.cs
i @../MyRND/MyPeripherals/STM32L4_SPI.cs

i @../MyRND/MyPeripherals/STM32L4_GPIOPort.cs
EnsureTypeIsLoaded "Antmicro.Renode.Peripherals.GPIOPort.STM32L4_GPIOPort"
i @../MyRND/MyPeripherals/ninaW15.cs

i @../MyRND/MyPeripherals/STM32L4_RTC.cs
EnsureTypeIsLoaded "Antmicro.Renode.Peripherals.Timers.STM32L4_RTC"
i @../MyRND/MyPeripherals/STM32L4_RCC.cs

machine LoadPlatformDescription @../MyRND/REPL/stm32l4_ninaW151.repl
sysbus LoadELF @../MyRND/ELF/bruxless-dock-fw.elf
showAnalyzer sysbus.usart2
start
```

`ninaW15` includes `STM32L4_GPIOPort`, in the same way as `STM32L4_RCC` does it with `STM32L4_RTC`.
I want to understand why we need to add both `EnsureTypeIsLoaded` calls to make my Renode simulation working. If I remove them, `STM32L4_GPIOPort` and `STM32L4_RTC` won't be found.

Thanks in advance.
Seb

PiotrZierhoffer, 2024-11-28:
Hi @spulverail 

`EnsureTypeIsLoaded` is required when a loaded type references another loaded type.

Each `i *.cs` creates a new assembly and Renode loads required assemblies lazily. 

However, the compilation process requires the type to be loaded at runtime, and because the compiler is a black box - we do not know which types will be required in a programmatic way.

`EnsureTypeIsLoaded` forces this load to happen immediately. We may change this behavior in a fairly near future, so that this command is not needed (it will not break anything though if used).

Hope that makes it clear!

spulverail, 2024-12-02:
Hi @PiotrZierhoffer ,
Thanks for the explanation, everything is perfectly clear!

closed 2024-12-02 by spulverail