renode/renode #713: How can i make my Custom model , C# code

godveryday, 2024-11-29:
If you declare Memory.MappedMemory Type in renode, it seems to be defined based on the C# code in infrastructure, but I can't find the source file in my Linux environment. I want to change that, where can I do it?

PiotrZierhoffer, 2024-11-30:
Hi @godveryday 

If you clone the repository, you need to initialize submodules - `git submodule update --init --recursive`.

This will give you the files, among them https://github.com/renode/renode-infrastructure/blob/master/src/Emulator/Main/Peripherals/Memory/MappedMemory.cs

godveryday, 2024-12-02:
I can't git clone using the company network, but I can install with a zip file. What should I do?
When I first installed the portable version, the problem was that the source code file was not visible.

PiotrZierhoffer, 2024-12-02:
You can go to https://builds.renode.io/ and download a `*source.tar.xz` file with the version matching your portable package - it should contain everything