renode/renode #717: failed tun ./Renode after build from source with --net --no-gui

iysheng, 2024-12-05:
Thanks for taking the time to file an issue for Renode!

Please familiarize yourself with our [Contributing guidelines](https://github.com/renode/renode/blob/master/CONTRIBUTING.md) first.

### Description

I build renode with command ``./build.sh --net --no-gui`` on fedora 40 success.

and the Renode version as:
``` bash
▸ ./Renode --version
Renode v1.15.3.41449
  build: 957fb8e6-202412052257
  build type: Release
  runtime: .NET 9.0.0
```
 But when I run command ./Renode, it failed and get me some errors like:

```
23:08:21.2262 [WARNING] Couldn't start UI - falling back to console mode
Fatal error:
Type doesn't have a method with a given name and signature. (Parameter 'type')
   at Microsoft.Scripting.Utils.ContractUtils.Requires(Boolean precondition, String paramName, String message)
   at Microsoft.Scripting.Generation.ILGen.EmitCall(Type type, String name, Type[] paramTypes)
   at IronPython.Runtime.Types.NewTypeMaker.ImplementCTDOverride(MethodInfo m)
   at IronPython.Runtime.Types.NewTypeMaker.ImplementCustomTypeDescriptor()
   at IronPython.Runtime.Types.NewTypeMaker.ImplementPythonObject()
   at IronPython.Runtime.Types.NewTypeMaker.ImplementType()
   at IronPython.Runtime.Types.NewTypeMaker.CreateNewType()
   at IronPython.Runtime.Types.NewTypeMaker.<>c__DisplayClass24_0.<GetNewType>b__0()
   at Microsoft.Scripting.Utils.Publisher`2.GetOrCreateValue(TKey key, Func`1 create)
   at IronPython.Runtime.Types.NewTypeMaker.GetNewType(String typeName, PythonTuple bases)
   at IronPython.Runtime.Types.PythonType.InitializeUserType(CodeContext context, String name, PythonTuple bases, PythonDictionary vars, String selfNames)
   at IronPython.Runtime.Types.PythonType..ctor(CodeContext context, String name, PythonTuple bases, PythonDictionary dict, String selfNames)
   at IronPython.Runtime.Types.PythonType.__new__(CodeContext context, PythonType cls, String name, PythonTuple bases, PythonDictionary dict, String selfNames)
   at IronPython.Runtime.Types.PythonType.__new__(CodeContext context, PythonType cls, String name, PythonTuple bases, PythonDictionary dict)
   at Microsoft.Scripting.Interpreter.FuncCallInstruction`6.Run(InterpretedFrame frame)
   at Microsoft.Scripting.Interpreter.Interpreter.Run(InterpretedFrame frame)
   at Microsoft.Scripting.Interpreter.LightLambda.Run6[T0,T1,T2,T3,T4,T5,TRet](T0 arg0, T1 arg1, T2 arg2, T3 arg3, T4 arg4, T5 arg5)
   at System.Dynamic.UpdateDelegates.UpdateAndExecute5[T0,T1,T2,T3,T4,TRet](CallSite site, T0 arg0, T1 arg1, T2 arg2, T3 arg3, T4 arg4)
   at Microsoft.Scripting.Interpreter.DynamicInstruction`6.Run(InterpretedFrame frame)
   at Microsoft.Scripting.Interpreter.Interpreter.Run(InterpretedFrame frame)
   at Microsoft.Scripting.Interpreter.LightLambda.Run2[T0,T1,TRet](T0 arg0, T1 arg1)
   at IronPython.Compiler.PythonScriptCode.RunWorker(CodeContext ctx)
   at IronPython.Compiler.PythonScriptCode.Run(Scope scope)
   at IronPython.Compiler.RuntimeScriptCode.InvokeTarget(Scope scope)
   at IronPython.Compiler.RuntimeScriptCode.Run(Scope scope)
   at Microsoft.Scripting.Hosting.CompiledCode.Execute(ScriptScope scope)
   at Antmicro.Renode.UserInterface.MonitorPythonEngine..ctor(Monitor monitor) in /home/yjoy/gProjects/renode/src/Infrastructure/src/Emulator/Extensions/UserInterface/MonitorPythonEngine.cs:line 39
   at Antmicro.Renode.UserInterface.Monitor..ctor() in /home/yjoy/gProjects/renode/src/Infrastructure/src/Emulator/Extensions/UserInterface/Monitor.cs:line 88
   at Antmicro.Renode.UI.CommandLineInterface.Run(Options options, Action`1 beforeRun) in /home/yjoy/gProjects/renode/src/Infrastructure/src/UI/CommandLineInterface.cs:line 86
   at Antmicro.Renode.Program.<>c__DisplayClass0_0.<Main>b__1() in /home/yjoy/gProjects/renode/src/Renode/Program.cs:line 54

Unhandled exception. System.ArgumentException: Type doesn't have a method with a given name and signature. (Parameter 'type')
   at Microsoft.Scripting.Utils.ContractUtils.Requires(Boolean precondition, String paramName, String message)
   at Microsoft.Scripting.Generation.ILGen.EmitCall(Type type, String name, Type[] paramTypes)
   at IronPython.Runtime.Types.NewTypeMaker.ImplementCTDOverride(MethodInfo m)
   at IronPython.Runtime.Types.NewTypeMaker.ImplementCustomTypeDescriptor()
   at IronPython.Runtime.Types.NewTypeMaker.ImplementPythonObject()
   at IronPython.Runtime.Types.NewTypeMaker.ImplementType()
   at IronPython.Runtime.Types.NewTypeMaker.CreateNewType()
   at IronPython.Runtime.Types.NewTypeMaker.<>c__DisplayClass24_0.<GetNewType>b__0()
   at Microsoft.Scripting.Utils.Publisher`2.GetOrCreateValue(TKey key, Func`1 create)
   at IronPython.Runtime.Types.NewTypeMaker.GetNewType(String typeName, PythonTuple bases)
   at IronPython.Runtime.Types.PythonType.InitializeUserType(CodeContext context, String name, PythonTuple bases, PythonDictionary vars, String selfNames)
   at IronPython.Runtime.Types.PythonType..ctor(CodeContext context, String name, PythonTuple bases, PythonDictionary dict, String selfNames)
   at IronPython.Runtime.Types.PythonType.__new__(CodeContext context, PythonType cls, String name, PythonTuple bases, PythonDictionary dict, String selfNames)
   at IronPython.Runtime.Types.PythonType.__new__(CodeContext context, PythonType cls, String name, PythonTuple bases, PythonDictionary dict)
   at Microsoft.Scripting.Interpreter.FuncCallInstruction`6.Run(InterpretedFrame frame)
   at Microsoft.Scripting.Interpreter.Interpreter.Run(InterpretedFrame frame)
   at Microsoft.Scripting.Interpreter.LightLambda.Run6[T0,T1,T2,T3,T4,T5,TRet](T0 arg0, T1 arg1, T2 arg2, T3 arg3, T4 arg4, T5 arg5)
   at System.Dynamic.UpdateDelegates.UpdateAndExecute5[T0,T1,T2,T3,T4,TRet](CallSite site, T0 arg0, T1 arg1, T2 arg2, T3 arg3, T4 arg4)
   at Microsoft.Scripting.Interpreter.DynamicInstruction`6.Run(InterpretedFrame frame)
   at Microsoft.Scripting.Interpreter.Interpreter.Run(InterpretedFrame frame)
   at Microsoft.Scripting.Interpreter.LightLambda.Run2[T0,T1,TRet](T0 arg0, T1 arg1)
   at IronPython.Compiler.PythonScriptCode.RunWorker(CodeContext ctx)
   at IronPython.Compiler.PythonScriptCode.Run(Scope scope)
   at IronPython.Compiler.RuntimeScriptCode.InvokeTarget(Scope scope)
   at IronPython.Compiler.RuntimeScriptCode.Run(Scope scope)
   at Microsoft.Scripting.Hosting.CompiledCode.Execute(ScriptScope scope)
   at Antmicro.Renode.UserInterface.MonitorPythonEngine..ctor(Monitor monitor) in /home/yjoy/gProjects/renode/src/Infrastructure/src/Emulator/Extensions/UserInterface/MonitorPythonEngine.cs:line 39
   at Antmicro.Renode.UserInterface.Monitor..ctor() in /home/yjoy/gProjects/renode/src/Infrastructure/src/Emulator/Extensions/UserInterface/Monitor.cs:line 88
   at Antmicro.Renode.UI.CommandLineInterface.Run(Options options, Action`1 beforeRun) in /home/yjoy/gProjects/renode/src/Infrastructure/src/UI/CommandLineInterface.cs:line 86
   at Antmicro.Renode.Program.<>c__DisplayClass0_0.<Main>b__1() in /home/yjoy/gProjects/renode/src/Renode/Program.cs:line 54

```

### Expected behaviour

run success.

### How to reproduce?

1. build renode with args ``--net --no-gui``
2. run ``./output/bin/Release/net6.0/Renode``

### Environment

Please, provide the following information:

- fedora 40
- Renode v1.15.3.41449   build: 957fb8e6-202412052257

### Additional information

I use ssh remote the PC build renode, and run Renode both is ssh remote

### Do you plan to address this issue and file a PR?

I need support.

PiotrZierhoffer, 2024-12-06:
I see the following:

```
10:17:23.0117 [WARNING] Couldn't start UI - falling back to console mode
10:17:23.7491 [INFO] Loaded monitor commands from: {...}renode/scripts/monitor.py
Renode, version 1.15.3.18418 (69a60d27-202412061013)

(monitor)
```

My best bet is that maybe you have an unclean build? Did you try to run `./build.sh -c` and then rebuild?

You can enforce a console-based run with:
```
./renode --net --disable-gui --console
```
or
```
./renode --net --disable-gui
```
if you'd like to connect via telnet to the Monitor.

This should, however, happen automatically if you build with `--no-gui`

iysheng, 2024-12-06:
> I see the following:
>
> ```
> 10:17:23.0117 [WARNING] Couldn't start UI - falling back to console mode
> 10:17:23.7491 [INFO] Loaded monitor commands from: {...}renode/scripts/monitor.py
> Renode, version [...]

after I do ./build.sh -c , then rebuild with ``./build.sh --no-gui``, after I build , then I do command with ``./renode --disable-gui``, I get error as
```
▸ ./renode --disable-gui
Fatal error:
Type doesn't have a method with a given name and signature. (Parameter 'type')                                                                                                                                      at Microsoft.Scripting.Utils.ContractUtils.Requires(Boolean precondition, String paramName, String message)
   at Microsoft.Scripting.Generation.ILGen.EmitCall(Type type, String name, Type[] paramTypes)
   at IronPython.Runtime.Types.NewTypeMaker.ImplementCTDOverride(MethodInfo m)
   at IronPython.Runtime.Types.NewTypeMaker.ImplementCustomTypeDescriptor()
   at IronPython.Runtime.Types.NewTypeMaker.ImplementPythonObject()
   at IronPython.Runtime.Types.NewTypeMaker.ImplementType()
   at IronPython.Runtime.Types.NewTypeMaker.CreateNewType()
   at IronPython.Runtime.Types.NewTypeMaker.<>c__DisplayClass24_0.<GetNewType>b__0()
   at Microsoft.Scripting.Utils.Publisher`2.GetOrCreateValue(TKey key, Func`1 create)
   at IronPython.Runtime.Types.NewTypeMaker.GetNewType(String typeName, PythonTuple bases)
   at IronPython.Runtime.Types.PythonType.InitializeUserType(CodeContext context, String name, PythonTuple bases, PythonDictionary vars, String selfNames)
   at IronPython.Runtime.Types.PythonType..ctor(CodeContext context, String name, PythonTuple bases, PythonDictionary dict, String selfNames)
   at IronPython.Runtime.Types.PythonType.__new__(CodeContext context, PythonType cls, String name, PythonTuple bases, PythonDictionary dict, String selfNames)
   at IronPython.Runtime.Types.PythonType.__new__(CodeContext context, PythonType cls, String name, PythonTuple bases, PythonDictionary dict)
   at Microsoft.Scripting.Interpreter.FuncCallInstruction`6.Run(InterpretedFrame frame)
   at Microsoft.Scripting.Interpreter.Interpreter.Run(InterpretedFrame frame)
   at Microsoft.Scripting.Interpreter.LightLambda.Run6[T0,T1,T2,T3,T4,T5,TRet](T0 arg0, T1 arg1, T2 arg2, T3 arg3, T4 arg4, T5 arg5)
   at System.Dynamic.UpdateDelegates.UpdateAndExecute5[T0,T1,T2,T3,T4,TRet](CallSite site, T0 arg0, T1 arg1, T2 arg2, T3 arg3, T4 arg4)
   at Microsoft.Scripting.Interpreter.DynamicInstruction`6.Run(InterpretedFrame frame)
   at Microsoft.Scripting.Interpreter.Interpreter.Run(InterpretedFrame frame)
   at Microsoft.Scripting.Interpreter.LightLambda.Run2[T0,T1,TRet](T0 arg0, T1 arg1)
   at IronPython.Compiler.PythonScriptCode.RunWorker(CodeContext ctx)
   at IronPython.Compiler.PythonScriptCode.Run(Scope scope)
   at IronPython.Compiler.RuntimeScriptCode.InvokeTarget(Scope scope)
   at IronPython.Compiler.RuntimeScriptCode.Run(Scope scope)
   at Microsoft.Scripting.Hosting.CompiledCode.Execute(ScriptScope scope)
   at Antmicro.Renode.UserInterface.MonitorPythonEngine..ctor(Monitor monitor)
   at Antmicro.Renode.UserInterface.Monitor..ctor()
   at Antmicro.Renode.UI.CommandLineInterface.Run(Options options, Action`1 beforeRun)
   at Antmicro.Renode.Program.<>c__DisplayClass0_0.<Main>b__1()

Unhandled exception. System.ArgumentException: Type doesn't have a method with a given name and signature. (Parameter 'type')
   at Microsoft.Scripting.Utils.ContractUtils.Requires(Boolean precondition, String paramName, String message)
   at Microsoft.Scripting.Generation.ILGen.EmitCall(Type type, String name, Type[] paramTypes)
   at IronPython.Runtime.Types.NewTypeMaker.ImplementCTDOverride(MethodInfo m)
   at IronPython.Runtime.Types.NewTypeMaker.ImplementCustomTypeDescriptor()
   at IronPython.Runtime.Types.NewTypeMaker.ImplementPythonObject()
   at IronPython.Runtime.Types.NewTypeMaker.ImplementType()
   at IronPython.Runtime.Types.NewTypeMaker.CreateNewType()
   at IronPython.Runtime.Types.NewTypeMaker.<>c__DisplayClass24_0.<GetNewType>b__0()
   at Microsoft.Scripting.Utils.Publisher`2.GetOrCreateValue(TKey key, Func`1 create)
   at IronPython.Runtime.Types.NewTypeMaker.GetNewType(String typeName, PythonTuple bases)
   at IronPython.Runtime.Types.PythonType.InitializeUserType(CodeContext context, String name, PythonTuple bases, PythonDictionary vars, String selfNames)
   at IronPython.Runtime.Types.PythonType..ctor(CodeContext context, String name, PythonTuple bases, PythonDictionary dict, String selfNames)
   at IronPython.Runtime.Types.PythonType.__new__(CodeContext context, PythonType cls, String name, PythonTuple bases, PythonDictionary dict, String selfNames)
   at IronPython.Runtime.Types.PythonType.__new__(CodeContext context, PythonType cls, String name, PythonTuple bases, PythonDictionary dict)
   at Microsoft.Scripting.Interpreter.FuncCallInstruction`6.Run(InterpretedFrame frame)
   at Microsoft.Scripting.Interpreter.Interpreter.Run(InterpretedFrame frame)
   at Microsoft.Scripting.Interpreter.LightLambda.Run6[T0,T1,T2,T3,T4,T5,TRet](T0 arg0, T1 arg1, T2 arg2, T3 arg3, T4 arg4, T5 arg5)
   at System.Dynamic.UpdateDelegates.UpdateAndExecute5[T0,T1,T2,T3,T4,TRet](CallSite site, T0 arg0, T1 arg1, T2 arg2, T3 arg3, T4 arg4)
   at Microsoft.Scripting.Interpreter.DynamicInstruction`6.Run(InterpretedFrame frame)
   at Microsoft.Scripting.Interpreter.Interpreter.Run(InterpretedFrame frame)
[... cut ...]
   at IronPython.Compiler.RuntimeScriptCode.Run(Scope scope)
   at Microsoft.Scripting.Hosting.CompiledCode.Execute(ScriptScope scope)
   at Antmicro.Renode.UserInterface.MonitorPythonEngine..ctor(Monitor monitor)
   at Antmicro.Renode.UserInterface.Monitor..ctor()
   at Antmicro.Renode.UI.CommandLineInterface.Run(Options options, Action`1 beforeRun)
   at Antmicro.Renode.Program.<>c__DisplayClass0_0.<Main>b__1()
./renode: line 11: 197979 Aborted                 (core dumped) $LAUNCHER $(dirname "${BASH_SOURCE[0]}")/bin/Renode.dll "$@"
┏
```

iysheng, 2024-12-06:
besides if I build with command ``./build.sh -p`` with mono, I will get failed message as
```
Core.dll /warn:4                                                                                                                                                                                                 CSC: error CS0518: The predefined type `System.Object' is not defined or imported                                                                                                                                CSC: error CS0518: The predefined type `System.ValueType' is not defined or imported                                                                                                                             CSC: error CS0518: The predefined type `System.Attribute' is not defined or imported                                                                                                                             CSC: error CS0518: The predefined type `System.Int32' is not defined or imported                                                                                                                                 CSC: error CS0518: The predefined type `System.UInt32' is not defined or imported                                                                                                                                CSC: error CS0518: The predefined type `System.Int64' is not defined or imported                                                                                                                                 CSC: error CS0518: The predefined type `System.UInt64' is not defined or imported                                                                                                                                CSC: error CS0518: The predefined type `System.Single' is not defined or imported                                                                                                                                CSC: error CS0518: The predefined type `System.Double' is not defined or imported                                                                                                                                CSC: error CS0518: The predefined type `System.Char' is not defined or imported                                                                                                                                  CSC: error CS0518: The predefined type `System.Int16' is not defined or imported                                                                                                                                 CSC: error CS0518: The predefined type `System.Decimal' is not defined or imported                                                                                                                               CSC: error CS0518: The predefined type `System.Boolean' is not defined or imported                                                                                                                               CSC: error CS0518: The predefined type `System.SByte' is not defined or imported                                                                                                                                 CSC: error CS0518: The predefined type `System.Byte' is not defined or imported                                                                                                                                  CSC: error CS0518: The predefined type `System.UInt16' is not defined or imported                                                                                                                                CSC: error CS0518: The predefined type `System.String' is not defined or imported                                                                                                                                CSC: error CS0518: The predefined type `System.Enum' is not defined or imported                                                                                                                                  CSC: error CS0518: The predefined type `System.Delegate' is not defined or imported                                                                                                                              CSC: error CS0518: The predefined type `System.MulticastDelegate' is not defined or imported                                                                                                                     CSC: error CS0518: The predefined type `System.Void' is not defined or imported                                                                                                                                  CSC: error CS0518: The predefined type `System.Array' is not defined or imported                                                                                                                                 CSC: error CS0518: The predefined type `System.Type' is not defined or imported                                                                                                                                  CSC: error CS0518: The predefined type `System.Collections.IEnumerator' is not defined or imported                                                                                                               CSC: error CS0518: The predefined type `System.Collections.IEnumerable' is not defined or imported                                                                                    [... cut ...]                                                                 Done building project "/home/yjoy/gProjects/renode/Renode.sln".-- FAILED                                                                                                                                                                                                                                                                                                                                                          Build FAILED. 
```

LillyJinstrand, 2024-12-06:
It seems that the original issue is caused by using the .NET 9 sdk. Not sure why it does not work currently, but for now if you can downgrade to .NET 8 sdk and that should resolve your issue

iysheng, 2024-12-06:
> It seems that the original issue is caused by using the .NET 9 sdk. Not sure why it does not work currently, but for now if you can downgrade to .NET 8 sdk and that should resolve your issue

I rebuild renode with mono, and it success. But when I do command ``showAnalyzer sysbus.usart1``, how could I see the new terminal ? PS, I use ssh to connect the PC installed renode.

PiotrZierhoffer, 2024-12-06:
In general, you should see it automatically, in console mode the UART will feed data to the log (you can verify that with `analyzers sysbus.usart1`, it will indicate that `LoggingUartAnalyzer` is the default one.

You can also use `uart_connect` https://renode.readthedocs.io/en/latest/basic/running.html#uart-interactions-in-the-monitor

Or any of these options: https://renode.readthedocs.io/en/latest/host-integration/uart.html

iysheng, 2024-12-07:
> In general, you should see it automatically, in console mode the UART will feed data to the log (you can verify that with `analyzers sysbus.usart1`, it will indicate that `LoggingUartAnalyzer` is [...]

After I launch renode with command ``./renode --console --hide-log``, then with ``uart_connect sysbus.usart1`` to redirecte the input to sysbus.usart1, I can use this console success, thanks for your advice.

iysheng, 2024-12-07:
> It seems that the original issue is caused by using the .NET 9 sdk. Not sure why it does not work currently, but for now if you can downgrade to .NET 8 sdk and that should resolve your issue

I use mono to build renode success, I wonder .Net or mono which is more better with renode . is there some compare ?

PiotrZierhoffer, 2024-12-10:
In general we promote dotnet - it's faster and mono support will be dropped in foreseeable future.

Right now we support dotnet 6, with dotnet 8 support coming in a few days. We also plan to relax our requirements to support whichever version the user chooses to uise

iysheng, 2024-12-10:
> In general we promote dotnet - it's faster and mono support will be dropped in foreseeable future.
> 
> Right now we support dotnet 6, with dotnet 8 support coming in a few days. We also plan to relax our requirements to support whichever version the user chooses to uise

thanks for your paient reply. 😊

closed 2024-12-10 by iysheng

jan-malek, 2024-12-10:
Hi. Just so you know as of commit d7fbd4d Renode targets dotnet 8 by default.

We also introduced a new flag `-F` to the `build.sh` script that allows you to specify which dotnet version should be used during the build. For example `./build.sh --net -F net6.0` can be used to build for dotnet 6.

Please note that this will not work for every dotnet version and this flag should be treated as an advanced build option.

DawidPietrykowski, 2025-01-23:
Just to note in case someone encounters the original issue caused by running Renode on .NET 9.0:
```
Unhandled exception. System.ArgumentException: Type doesn't have a method with a given name and signature. (Parameter 'type')
   at Microsoft.Scripting.Utils.ContractUtils.Requires(Boolean precondition, String paramName, String message)
   at Microsoft.Scripting.Generation.ILGen.EmitCall(Type type, String name, Type[] paramTypes)
   at IronPython.Runtime.Types.NewTypeMaker.ImplementCTDOverride(MethodInfo m)
   at IronPython.Runtime.Types.NewTypeMaker.ImplementCustomTypeDescriptor()
   at IronPython.Runtime.Types.NewTypeMaker.ImplementPythonObject()
   at IronPython.Runtime.Types.NewTypeMaker.ImplementType()
   at IronPython.Runtime.Types.NewTypeMaker.CreateNewType()
   at IronPython.Runtime.Types.NewTypeMaker.<>c__DisplayClass24_0.<GetNewType>b__0()
```

It has been fixed in 34b3a9e346b255da101b664d0ce0404b06f4d96d and you can now run Renode on .NET 9.0