renode/renode #726: Verilator cosimulation libraries cannot be built from non-source installation

Arusekk, 2025-01-09:
I tried to build some co-simulation libraries according to the instructions in https://github.com/antmicro/renode-verilator-integration, but I used portable installation (and pointed RENODE_ROOT there, actually by `cmake -DCOSIM_DIR`) instead of one built from source.
The culprit was the following line:

https://github.com/renode/renode/blob/606a24e00955cc8338bdbee47cf32140adf965bb/src/Plugins/CoSimulationPlugin/IntegrationLibrary/src/renode_bus.h#L11

I managed to work around the issue by copying both renode_imports.h and renode_imports_generated.h from source to the same directory, and adjusting the `../../../..` line for that.

­

(Side note: I found an arguably “simpler” way to compile the ibex CPU without using CMake:)
<details>

```sh
sed s/ibex_pkg/ibex_top/ sim_ibex.h
mkdir build && cd build
fusesoc --cores-root=../ibex run --target=default --tool verilator --setup lowrisc:ibex:ibex_top
cd build/lowrisc_ibex_ibex_top_0.1/default-verilator
verilator \
    --lib-create Vcpu_ibex --build \
    $(grep -v -- --exe lowrisc_ibex_ibex_top_0.1.vc) \
    ../../../sim_main.cpp ../../../sim_ibex.cpp \
    $(find $RENODE_ROOT/plugins/VerilatorIntegrationLibrary -name '*.cpp' \
         |grep -v renode_cfu |grep -v buses) \
    -CFLAGS -I$RENODE_ROOT/plugins/VerilatorIntegrationLibrary
```
</details>

JanOlencki, 2025-01-09:
Hi @Arusekk
You are right that at the moment the portable version of Renode isn't enough to build an example from the `renode-verilator-integration` repository. You need to clone the Renode repository as described in our documentation. You don't need to build Renode from source, just make the source available for CMake.
Keep in mind that the [DPI integration](https://github.com/antmicro/renode-dpi-examples) doesn't have this requirement, you can treat it as an alternative approach to co-simulation that relies only on the files available in portable Renode.

Arusekk, 2026-10-30:
Fixed by c657bbf8a0b0cb645ffd118767207282869bc3bf

closed 2026-10-30 by Arusekk