renode/renode #729: Support Needed for Running Renode from `.pkg.tar.xz` on Windows

akhilender-bongirwar, 2025-01-15:
## Description  
Hello,  

I am a student developer exploring Renode and working on creating a single package that can run seamlessly on all major operating systems. To achieve this, I downloaded Renode from [this release link](https://github.com/renode/renode/releases/download/v1.15.3/renode-1.15.3-1-x86_64.pkg.tar.xz) and tested it successfully on Linux and macOS.  

On these platforms, I was able to navigate to `/opt/renode/bin` and execute `mono Renode.exe`, which worked perfectly.  

However, when trying to set it up on Windows, I encountered issues. Specifically:  
- After navigating to `/opt/renode/bin`, I tried executing:  
  - `dotnet Renode-peripherals.dll`  
- The command resulted in errors, and I was unable to proceed further.  

I understand from [this reply](https://builds.renode.io/) that the preferred method on Windows is using the portable package along with Cygwin. However, I would appreciate guidance or support to understand if it's feasible to use the `.pkg.tar.xz` package directly on Windows through dotnet, as I aim for cross-platform compatibility with a single package.  

Could you please provide insights or instructions for getting this to work on Windows?  

## Steps to Reproduce  
1. Download Latest Renode package: [renode-1.15.3-1-x86_64.pkg.tar.xz](https://github.com/renode/renode/releases/download/v1.15.3/renode-1.15.3-1-x86_64.pkg.tar.xz).  
2. Extract and navigate to `/opt/renode/bin`.  
3. Run `dotnet Renode-peripherals.dll`.  

## Expected Behavior  
Renode starts successfully on Windows using the `.pkg.tar.xz` package.  

## Actual Behavior  
Dotnet commands result in errors, and Renode does not start.  

## Additional Information  
- **OS**: Windows 10/11  
- **Dotnet version**: 7.0.306 [C:\Program Files\dotnet\sdk]  

**PS:** I am new to using dotnet, so please let me know if the command `dotnet Renode-peripherals.dll` is correct or if  I might have missed something important from the docs for windows. @WojMel,  thank you for your advice to open an issue.

WojMel, 2025-01-21:
Hi @akhilender-bongirwar
The `.pkg.tar.xz` is a mono build of Renode for Linux. To run dotnet version of Renode on Windows use [renode-latest.windows-portable-dotnet.zip](https://builds.renode.io/renode-latest.windows-portable-dotnet.zip) and start the included `renode.exe` executable.

akhilender-bongirwar, 2025-01-21:
> Hi [@akhilender-bongirwar](https://github.com/akhilender-bongirwar) The `.pkg.tar.xz` is a mono build of Renode for Linux. To run dotnet version of Renode on Windows use [...]

I understand your point, but my goal is to achieve cross-platform compatibility of Renode using a single package. The solution you suggested would work well for Windows but is limited to that platform. I am exploring ways to make the `.pkg.tar.xz` package universally compatible across all operating systems. 

PS: @pkoscik tagging for reference, thanks for replying back on email.