renode/renode #730: ST Nucleo WB55RG Nucleo Base Support

toastbucket, 2025-01-17:
### Description

Basic support for Nucleo WB55RG board.
- `.repl` file for stm32wb55
- `.repl` file for stm32wb55 nucleo
- `.resc` file for stm32wb55 nucleo
- Enough peripheral support for Zephyr basic blinky

### Usage example

`machine LoadPlatformDescription @platforms/boards/nucleo_wb55rg`

### Additional information

I have this working in my fork with some light modifications to existing peripherals. I need to clean it up and provide proper WB55 peripherals where appropriate.

### Do you plan to address this issue and file a PR?

I plan to work on this

nickelsmore-philips, 2025-01-31:
I have my implementation cleaned up and ready for a PR.  However, I'm having a difficult time understanding the desired approach using the Issue Reproduction Template.  I'm hoping a maintainer can help guide me in getting this code properly integrated.

I have it currently on my personal fork: https://github.com/toastbucket/renode/tree/730-stm32wb55_support

toastbucket, 2025-01-31:
I have my implementation cleaned up and ready for a PR.  However, I'm having a difficult time understanding the desired approach using the Issue Reproduction Template.  I'm hoping a maintainer can help guide me in getting this code properly integrated.

I have it currently on my personal fork: https://github.com/toastbucket/renode/tree/730-stm32wb55_support