renode/renode #731: Peripheral Device Support Single-wire Serial Interface

JohnZhang3IoT, 2025-01-20:
Hi guys, I know in renode peripheral device simulation, it is easy to simulate the I2C and SPI device using C# model, but now I want to know how to simulate single-wire interface sensor? These sensor using single GPIO to communicate, and their protocol is timing-relevant, where you need read the pulse within certain period of time, so how to simulate this logic, do you have any idea?

DHT11 is one of the temperature/humidity sensor using single-wire serial interface, you may find reference from the datasheet[https://www.mouser.com/datasheet/2/758/DHT11-Technical-Data-Sheet-Translated-Version-1143054.pdf?srsltid=AfmBOop-OcM7ZJfMx1QkOxnUtLze8Wv8EjsZx1czQlPb6ILgFo2X0YWI](url) 

I am trying to use machine.ElapsedVirtualTime for handle time information, but I am not sure if it is the correct direction
             
```
if (machine != null)
 {
      currentTime = machine.ElapsedVirtualTime;
      var duration = currentTime.TimeElapsed - lastTime.TimeElapsed;
      this.Log(LogLevel.Info, $"LED state changed to {state} for {duration.TotalMilliseconds} ms.");
      lastTime = currentTime;
  }     
```

JohnZhang3IoT, 2025-01-21:
I write a DHT11 C# model, because it need bahave as both input and output devices for command and data, so I did this 
`public class DHT11 : IPeripheral, IGPIOReceiver, IGPIOSender` 
to support both input and output.
However, when I start this simulation, I got the following error, it seems if the model when implements both IGPIOReceiver, IGPIOSender, will have conflict and produce error. I attached **.resc .repl, .elf and DHT11.cs model here** [stm32f4_dht11.zip](https://github.com/user-attachments/files/18493815/stm32f4_dht11.zip), hope anybody could give some suggestions, thanks!

```
Object reference not set to an instance of an object.
   at Antmicro.Renode.PlatformDescription.Syntax.SyntaxTreeHelpers.<>c__DisplayClass0_0.<ScanFor>b__0(IWithPosition x) in C:\WorkSpace\Embedded\Renode\renode\src\Renode\PlatformDescription\Syntax\SyntaxTreeHelpers.cs:line 21
   at Antmicro.Renode.PlatformDescription.Syntax.SyntaxTreeHelpers.VisitSyntaxTreeInner[TValue](Object objectToVisit, Action`1 visitorAction, Func`3 filter) in C:\WorkSpace\Embedded\Renode\renode\src\Renode\PlatformDescription\Syntax\SyntaxTreeHelpers.cs:line 41
   at Antmicro.Renode.PlatformDescription.Syntax.SyntaxTreeHelpers.VisitSyntaxTreeInner[TValue](Object objectToVisit, Action`1 visitorAction, Func`3 filter) in C:\WorkSpace\Embedded\Renode\renode\src\Renode\PlatformDescription\Syntax\SyntaxTreeHelpers.cs:line 53
   at Antmicro.Renode.PlatformDescription.Syntax.SyntaxTreeHelpers.VisitSyntaxTree[TValue](Object root, Action`1 visitorAction) in C:\WorkSpace\Embedded\Renode\renode\src\Renode\PlatformDescription\Syntax\SyntaxTreeHelpers.cs:line 28
   at Antmicro.Renode.PlatformDescription.Syntax.SyntaxTreeHelpers.ScanFor(Object objectToScan, IWithPosition objectToFind) in C:\WorkSpace\Embedded\Renode\renode\src\Renode\PlatformDescription\Syntax\SyntaxTreeHelpers.cs:line 19
   at Antmicro.Renode.PlatformDescription.CreationDriver.GetElementSourceAndPath(IWithPosition element, String& file, String& source) in C:\WorkSpace\Embedded\Renode\renode\src\Renode\PlatformDescription\CreationDriver.cs:line 1680
   at Antmicro.Renode.PlatformDescription.CreationDriver.HandleError(ParsingError error, IWithPosition failingObject, String message, Boolean longMark) in C:\WorkSpace\Embedded\Renode\renode\src\Renode\PlatformDescription\CreationDriver.cs:line 1625
   at Antmicro.Renode.PlatformDescription.CreationDriver.ValidateEntryPreMerge(Entry entry) in C:\WorkSpace\Embedded\Renode\renode\src\Renode\PlatformDescription\CreationDriver.cs:line 497
   at Antmicro.Renode.PlatformDescription.CreationDriver.ValidateEntriesPreMerge(List`1 entries) in C:\WorkSpace\Embedded\Renode\renode\src\Renode\PlatformDescription\CreationDriver.cs:line 394
   at Antmicro.Renode.PlatformDescription.CreationDriver.ValidatePreMerge(String file, String source, String prefix) in C:\WorkSpace\Embedded\Renode\renode\src\Renode\PlatformDescription\CreationDriver.cs:line 163
   at Antmicro.Renode.PlatformDescription.CreationDriver.ProcessInner(String file, String source) in C:\WorkSpace\Embedded\Renode\renode\src\Renode\PlatformDescription\CreationDriver.cs:line 68
   at Antmicro.Renode.PlatformDescription.CreationDriver.ProcessFile(String path) in C:\WorkSpace\Embedded\Renode\renode\src\Renode\PlatformDescription\CreationDriver.cs:line 61
   at Antmicro.Renode.PlatformDescription.UserInterface.PlatformDescriptionMachineExtensions.LoadPlatformDescription(Machine machine, String platformDescriptionFile) in C:\WorkSpace\Embedded\Renode\renode\src\Renode\PlatformDescription\UserInterface\PlatformDescriptionMachineExtensions.cs:line 19
   at Dynamitey.Internal.Optimization.InvokeHelper.InvokeMemberAction(CallSite& callsite, Type binderType, Int32 knownType, LazyBinder binder, InvokeMemberName name, Boolean staticContext, Type context, String[] argNames, Object target, Object[] args)
   at Dynamitey.Internal.Optimization.InvokeHelper.InvokeMemberActionCallSite(Object target, InvokeMemberName name, Object[] args, String[] tArgNames, Type tContext, Boolean tStaticContext, CallSite& callSite)
   at Dynamitey.Dynamic.InvokeMemberAction(Object target, String_OR_InvokeMemberName name, Object[] args)
   at Antmicro.Renode.UserInterface.Monitor.InvokeWithContext(InvokeContext context, MethodInfo method, Object[] parameters) in C:\WorkSpace\Embedded\Renode\renode\src\Infrastructure\src\Emulator\Extensions\UserInterface\MonitorCommands.cs:line 1025
   at Antmicro.Renode.UserInterface.Monitor.InvokeExtensionMethod(Object device, MethodInfo method, List`1 parameters) in C:\WorkSpace\Embedded\Renode\renode\src\Infrastructure\src\Emulator\Extensions\UserInterface\MonitorCommands.cs:line 974
   at Antmicro.Renode.UserInterface.Monitor.ExecuteDeviceAction(String name, Object device, IEnumerable`1 p) in C:\WorkSpace\Embedded\Renode\renode\src\Infrastructure\src\Emulator\Extensions\UserInterface\MonitorCommands.cs:line 1295
   at Antmicro.Renode.UserInterface.Monitor.ProcessDeviceAction(Type deviceType, String name, IEnumerable`1 p, ICommandInteraction writer) in C:\WorkSpace\Embedded\Renode\renode\src\Infrastructure\src\Emulator\Extensions\UserInterface\MonitorCommands.cs:line 425
   at Antmicro.Renode.UserInterface.Monitor.ProcessDeviceActionByName(String name, IEnumerable`1 p, ICommandInteraction writer) in C:\WorkSpace\Embedded\Renode\renode\src\Infrastructure\src\Emulator\Extensions\UserInterface\MonitorCommands.cs:line 458
[... cut ...]
   at Antmicro.Renode.UI.CommandLineInterface.<>c__DisplayClass0_1.<Run>b__3(Object x) in C:\WorkSpace\Embedded\Renode\renode\src\Infrastructure\src\UI\CommandLineInterface.cs:line 107
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)
   at System.Threading.ThreadHelper.ThreadStart(Object obj)
```

naitgacem, 2025-02-01:
I was able to bypass the crash after adding a registration pin number like so:
`dht11: Miscellaneous.DHT11 @ gpioPortA 4`

But if we remove the IGPIOSender it works fine. Maybe you should replace it with a GPIO output like the IRQ ?
From the docs, I remember seeing that GPIO connections are asymetrical.

JohnZhang3IoT, 2025-02-02:
> I was able to bypass the crash after adding a registration pin number like so: `dht11: Miscellaneous.DHT11 @ gpioPortA 4`
>
> But if we remove the IGPIOSender it works fine. Maybe you should replace it [...]

Thanks @naitgacem , yes after removed IGPIOSender it works fine, I will modify the C# and try again; another though is, maybe can use second board to write firmware to simulate the sensor behavior