renode/renode #735: Support for Cortex M55 CPU in Renode

iiskcph, 2025-01-31:
Hi, I wanted to check if there is current support for the Cortex M55 CPU in Renode. If not, is there any plan to add support for the Cortex M55 CPU in the future?

### Usage example

I would like to know if such support would include both the CPU and memory mapping to the hardware. Is there a list or resource that specifies the correlation between supported boards and CPUs?

[image]

I appreciate any insights you can provide!

Thank you!

naitgacem, 2025-02-01:
I think Cortex-M is a model of its own. Here is an example:

[image]

As for memory mapping, do you mean peripheral mapping? In such case you can consult the datasheet and reconstruct the peripheral memory mapping of your particular device. If the peripherals are already modeled it is trivial to build a system from them.

---

I am not with the staff, just trying to help :P

dreambucket13, 2025-08-30:
To me the more relevant question is about Helium instructions - are they supported?