renode/renode #740: Failed to build on Windows

naitgacem, 2025-02-05:
### Description

I followed the instruction at https://renode.readthedocs.io/en/latest/advanced/building_from_sources.html and everything goes fine until the very last step of packaging. I get the following error:

```
Microsoft (R) Build Engine version 16.11.2+f32259642 for .NET Framework
Copyright (C) Microsoft Corporation. All rights reserved.

Build started 2/5/2025 5:41:34 PM.
Project "C:\Users\nabil\Downloads\renode\tools\packaging\windows\RenodeSetup\SetupProject.wixproj" on node 1 (Clean;Build target(s)).
CoreClean:
  Deleting file "obj\Release\_Items_dir.wxs".
  Deleting file "obj\Release\Product.wixobj".
  Deleting file "obj\Release\RenodeUI.wixobj".
  Deleting file "obj\Release\_Items_dir.wixobj".
HarvestDirectory:
  C:\Program Files (x86)\WiX Toolset v3.14\bin\Heat.exe dir ..\..\renode_1.15.3\ -cg Items -dr INSTALLDIR -sreg -srd -var var.ItemDir -ag -sfrag -out obj\Release\_Items_dir.wxs
Compile:
  C:\Program Files (x86)\WiX Toolset v3.14\bin\candle.exe -dVERSION=1.15.3 -dItemDir=..\..\renode_1.15.3\ -dConfiguration=Release -dOutDir=bin\Release\ -dPlatform=x64 -dProjectDir=C:\Users\na
  bil\Downloads\renode\tools\packaging\windows\RenodeSetup\ -dProjectExt=.wixproj -dProjectFileName=SetupProject.wixproj -dProjectName=SetupProject -dProjectPath=C:\Users\nabil\Downloads\reno
  de\tools\packaging\windows\RenodeSetup\SetupProject.wixproj -dTargetDir=C:\Users\nabil\Downloads\renode\tools\packaging\windows\RenodeSetup\bin\Release\ -dTargetExt=.msi -dTargetFileName=re
  node_1.15.3.msi -dTargetName=renode_1.15.3 -dTargetPath=C:\Users\nabil\Downloads\renode\tools\packaging\windows\RenodeSetup\bin\Release\renode_1.15.3.msi -out obj\Release\ -arch x64 -ext "C
  :\Program Files (x86)\WiX Toolset v3.14\bin\\WixUIExtension.dll" Product.wxs RenodeUI.wxs obj\Release\_Items_dir.wxs
Link:
  C:\Program Files (x86)\WiX Toolset v3.14\bin\Light.exe -out C:\Users\nabil\Downloads\renode\tools\packaging\windows\RenodeSetup\bin\Release\renode_1.15.3.msi -pdbout C:\Users\nabil\Download
  s\renode\tools\packaging\windows\RenodeSetup\bin\Release\renode_1.15.3.wixpdb -cultures:null -ext "C:\Program Files (x86)\WiX Toolset v3.14\bin\\WixUIExtension.dll" -contentsfile obj\Releas
  e\SetupProject.wixproj.BindContentsFileListnull.txt -outputsfile obj\Release\SetupProject.wixproj.BindOutputsFileListnull.txt -builtoutputsfile obj\Release\SetupProject.wixproj.BindBuiltOut
  putsFileListnull.txt -wixprojectfile C:\Users\nabil\Downloads\renode\tools\packaging\windows\RenodeSetup\SetupProject.wixproj obj\Release\Product.wixobj obj\Release\RenodeUI.wixobj obj\Rele
  ase\_Items_dir.wixobj
light.exe : error LGHT0001: The file cannot be accessed by the system. [C:\Users\nabil\Downloads\renode\tools\packaging\windows\RenodeSetup\SetupProject.wixproj]
Done Building Project "C:\Users\nabil\Downloads\renode\tools\packaging\windows\RenodeSetup\SetupProject.wixproj" (Clean;Build target(s)) -- FAILED.

Build FAILED.

"C:\Users\nabil\Downloads\renode\tools\packaging\windows\RenodeSetup\SetupProject.wixproj" (Clean;Build target) (1) ->
(Link target) ->
  light.exe : error LGHT0001: The file cannot be accessed by the system. [C:\Users\nabil\Downloads\renode\tools\packaging\windows\RenodeSetup\SetupProject.wixproj]

    0 Warning(s)
    1 Error(s)

Time Elapsed 00:00:03.38
```

### Expected behaviour

The build to produce an installer.

### How to reproduce?

Follow the instructions by the letter. The only modification is hardcoding the `ROOT_PATH` variable to allow the build to go forward.

### Environment

- Windows
- Git cloned into local directory

jeffglaum, 2025-02-19:
If you built under Windows using the cygwin instructions, the problem is likely this line in common_copy_files.sh:

ln -s ../../IntegrationLibrary/libs/socket-cpp $DIR/plugins/SystemCModule/lib/socket-cpp

The cygwin symlink looks like an NTFS junction to Windows and it seems that the WIX light/linker tool can't handle the junction.

Unclear why this may have worked for others - the symlink was first added in June of 2024.  First chance, I'll put up a PR...

naitgacem, 2025-02-21:
Hi again @jeffglaum. Thanks a lot! 

That did fix both issues. Nice work, closing now.

closed 2025-02-21 by naitgacem