renode/renode #742: Robot tests getting stuck

NicolaKeo, 2025-02-06:
### Description

Robot tests in one particular test suite are frequently getting stuck. Around 70% of the time they run successfully but the rest of the time the job will get stuck and have to be forcibly cancelled. 

I added extra logs to try identify the issue using `sysbus.cpu LogFunctionNames true` and `sysbus LogAllPeripheralsAccess true`.

I saw in one case there was an error with the program counter getting  corrupted during “context” switch from PendSV to LPUART interrupt.

[image]

I tried to do a memory dump using  `Execute Command  dump_file` in the test teardown stage but when the test has to be forcibly cancelled, the teardown stage is not executed.

Do you have an idea what is happening? 
Is there is some more steps we can do to debug this issue?

### Environment

- OS - Ubuntu 20.04
- Renode v1.15.3.4441
  build: 0a17b4a6-202501210324
  build type: Release
  runtime: .NET 8.0.12

tobermory, 2025-03-06:
I'm not familiar with the Renode code, but know the Cortex-M pretty well.  You don't say your CPU is a Cortex-M, but PendSV exists as a system exception on CM, and has isr num 14.

If the PendSV handler is running, the stack would be the Main Stack, i.e. using MSP.  If the PendSV handler is interrupted by your LPUART1 interrupt, which would have to have priority higher (numerically LOWER) than the PendSV handler, then 8 regs, including the PC, would be pushed to the main stack. Once the LPUart1 handler finishes, those 8 regs, inc PC, should be popped off the main stack into the regs, inc the PC in PendSV handler that was interrupted.

Now, your corrupted PC could be a Renode failure to implement the correct nested exceptions/interrupts logic.  But it could also be that your lpuart1 handler is itself flawed.  If it performed any 'stack smashing', i.e. wrote to the stack outside of its allocated 'stack frame', any of the 8 stacked regs that 'belong' to the PendSV handler could in theory be corrupted.

Does your code run perfectly every time on real hardware?

tobermory, 2025-03-06:
Hmm, I thought your PC corruption was when returning from LPUART1 handler to the PendSV handler.  Re-reading your post, I think you are suggesting that the corruption is actually happening when attempting to ENTER the LPUART1 handler??  That address should be loaded from Vectors[64], if 64 is the IRQn_Type value for your LPUART1 interrupt. 

As I said earlier, the Vectors table is at SCB->VTOR.  The SCB->VTOR is likely correct, since the PendSV handler PC location itself would have come from Vectors[14], and your PendSV handler was entered OK. Of course it's possibly, but v unlikely, that your PendSV handler CHANGED VTOR before it was interrupted by LPUART1.