renode/renode #755: Windows 11 builds under Cygwin

jeffglaum, 2025-02-21:
Thanks for taking the time to file an issue for Renode!

Please familiarize yourself with our [Contributing guidelines](https://github.com/renode/renode/blob/master/CONTRIBUTING.md) first.

### Description

Exeduting "./build.sh -p" under cygwin results in two issues:

1. cmake error:

CMake Error: The source directory "/cygdrive/c/r/renode/src/Infrastructure/src/Emulator/Cores" does not exist.

2. wix packaging error (once you get past #1):

  light.exe : error LGHT0001: The file cannot be accessed by the system. [C:\Users\nabil\Downloads\renode\tools\packaging\windows\RenodeSetup\SetupProject.wixproj]

### Expected behaviour

Expect to be able to build under Windows.

### How to reproduce?

Follow the build instructions for Windows as written: https://renode.readthedocs.io/en/latest/advanced/building_from_sources.html

### Environment

Please, provide the following information:

- OS: Windows 11 (24H2, OS Build 26100.3194)
- Renode version (with commit SHA): commit SHA 5932d2c410f13e3b10c905be3dbaa1d220774896

### Additional information

NA

### Do you plan to address this issue and file a PR?

I will put up a PR with the fixes.

Leeharim-korean, 2025-02-26:
I added some code in build.sh

line 309 : origin code
```
CORES_PATH="$ROOT_PATH/src/Infrastructure/src/Emulator/Cores"
```

New code
```
CORES_PATH="$ROOT_PATH/src/Infrastructure/src/Emulator/Cores"
if grep -q "CYGWIN" /proc/version
then
    CORES_PATH=$(cygpath -w "$CORES_PATH")
fi
```

LillyJinstrand, 2025-03-26:
Hi @jeffglaum, and @Leeharim-korean.
As of Renode commit [5c4382f](https://github.com/renode/renode/commit/5c4382fbc69b2701b8fbcc21c3526e8b92678958) we have updated the build flow on Windows to use Git Bash instead of Cygwin's shell. Updated instructions can be found [here](https://renode.readthedocs.io/en/latest/advanced/building_from_sources.html) but the short version is that Cygwin is no longer needed, and the build should be ran in Git Bash instead of Cygwin. If you have any issues with this flow please open an issue and I will look into it promptly

closed 2025-03-26 by LillyJinstrand